RROrg/rr #2: DS920的pcie10G网卡支持（TL-NT521）

Pan5094, 2023-04-19:
你好 我用arpl，918/920可硬解但是无法加载pcie 10g网卡，3622可加载10g网卡，但是无法硬解。可以920加载pcie 10G网卡嘛

Pan5094, 2023-04-19:
marvell v2这个驱动只有在白群晖有pcie扩展的型号上加载，918/920只加载marvell，不加载marvell v2

lianjianjie, 2023-04-20:
有类似的问题，DS920 x520万兆网卡同样无法驱动

Pan5094, 2023-04-20:
我的问题已解决，驱动是存在的，修改最大网卡数量即可！

CarlosLucky, 2023-04-24:
我的黑群晖主板原生1G网卡，我安装了一个2.5G PCIE tplink的网卡，现在随机出现重启没反应，或者上电开机没反应的情况，有时候正常，有时候重启就没任何反应包括关机状态下上电开机也是这样，在路由器端会发现每次出现此类情况的时候（我默认连接的是2.5G网口，路由器固定了网卡mac）路由器会显示1G网口已经连接实际上是2.5G网口连接，然后过段时间就消失连接了路由器上发现不了设备，我查看了arpl1分区下的user-config.yml文件cmdline设置为2个网卡mac1我配置了原生1G，mac2配置2.5G；同时original-mac也指向的是2.5G默认连接端口啊。出现这种情况后我必须手动重启一下主机，然后就能识别了。

wjz304, 2023-04-24:
> 我的黑群晖主板原生1G网卡，我安装了一个2.5G PCIE [...]

23.4.3么？

CarlosLucky, 2023-04-27:
是的，我目前需要升级成23.4.4试试么？

wjz304, 2023-04-29:
try the new version

wjz304, 2023-06-09:
原始问题已解决，如新版有问题，请重新创建issue

closed 2023-06-09 by wjz304