RROrg/rr #3: 引导异常屏蔽端口

SiegFish, 2023-04-20:
测试环境：esxi8.0 u1
系统版本：DSM7.2

引导会屏蔽第二个sata控制器的前两个端口（31，32），第三个控制器的前4个端口（61，62，63，64）。一般直通硬盘控制器的话是占用第二个sata控制器，屏蔽两块盘会导致raid5阵列直接损毁。

Pan5094, 2023-04-20:
d 

有关DiskIdxMap、SataPortMap、SasIdxMap三个参数的说明：

1、在群辉系统中，一个控制器默认只能控制9个硬盘，所以SataPortMap=的数值范围只能是0-9，0则表示屏蔽该控制器，不可以设为SataPortMap=0来隐藏引导盘！

2、如果有多个控制器的（每个主板具体有几个控制器，请查阅主板说明书），则SataPortMap=后面的数值有几位就表示有几个控制器。比如SataPortMap=148，表明第一个控制器有1个硬盘，第二个控制器有4个硬盘，第三个控制器有8个硬盘。

3、DiskIdxMap=这个参数值，按顺序从左到右每两位数为一个控制器的盘序数值，这两位数为16进制数，如果数值小于等于9或者大于10的，需要在数值前面加个0。比如DiskIdxMap=000105的意思是，第一个控制器的磁盘（引导盘）从序号1（00为序列1）开始往后排，第二个控制器的4个硬盘从序号2（01为序号2）开始往后排，第三个驱动器的8个硬盘从序号5（05为序号6）开始往后排。此时在群晖DSM7.X的系统里面看到磁盘序号就会从磁盘1（引导盘）到磁盘13。

4、这里有个小技巧，可以将控制器的盘序设置超过当前最大磁盘数量。例如我想把第1个控制器的引导盘隐藏，可以改成DiskIdxMap=0C0004，引导盘在群晖系统里面就看不到了，此时在群晖DSM7.X的系统里面看到磁盘序号就会从磁盘1到磁盘12（引导盘被隐藏了），同时还治愈了很多患有强迫症的童鞋！

5、SasIdxMap为SAS硬盘的个数，使用SATA硬盘的时候默认设置为0。

SiegFish, 2023-04-21:
> d
>
> 有关DiskIdxMap、SataPortMap、SasIdxMap三个参数的说明：
>
> 1、在群辉系统中，一个控制器默认只能控制9个硬盘，所以SataPortMap=的数值范围只能是0-9，0则表示屏蔽该控制器，不可以设为SataPortMap=0来隐藏引导盘！ [...]

> 

跟这个没有关系，APRL默认添加的参数是3个控制器每个30口，硬盘在引导里的show all sata里都可以读到，只是在群晖系统安装界面读不到对应端口的盘。并且硬盘位置也是对的，比如第三个控制器加5盘，前四个被屏蔽，安装界面读出的盘序号是65。

SiegFish, 2023-04-21:
> 跟这个没有关系，APRL默认添加的参数是3个控制器每个30口，硬盘在引导里的show all sata里都可以读到，只是在群晖系统安装界面读不到对应端口的盘。并且硬盘位置也是对的，比如第三个控制器加5盘，前四个被屏蔽，安装界面读出的盘序号是65。

我昨天试了7.2，7.1.1都有类似的问题，而原版arpl的7.1.1就可以正常识别所有盘。

wjz304, 2023-04-22:
卡8%的根本原因是 硬盘总是超过27个 raidtool 工具就会报错，
一般也只有 vmware/esxi 会是每个控制器 30个ata 。

当前是限制了最大硬盘数是26. 所以 导致可能会出现读不到的情况，

wjz304, 2023-04-22:
想到一种思路，或许能兼容两个问题，但是没有测试环境，用esxi的 能用那个就用哪个吧。

closed 2023-04-24 by SiegFish