RROrg/rr #4: ESXI 7.0 U3安装DS3622XS+ 直通硬盘识别不完整怎么解决？

xinjing-cloud, 2023-04-20:
4块直通硬盘，只能看到33和34号盘，35和36显示不出来，之前可以通过grup里面的参数修改硬盘位置，请教一下现在的版本如何修改硬盘位置？

wjz304, 2023-04-20:
我不知道你指的grup 是什么，设置菜单与原版并没有区别。

Pan5094, 2023-04-20:
d 

有关DiskIdxMap、SataPortMap、SasIdxMap三个参数的说明：

1、在群辉系统中，一个控制器默认只能控制9个硬盘，所以SataPortMap=的数值范围只能是0-9，0则表示屏蔽该控制器，不可以设为SataPortMap=0来隐藏引导盘！

2、如果有多个控制器的（每个主板具体有几个控制器，请查阅主板说明书），则SataPortMap=后面的数值有几位就表示有几个控制器。比如SataPortMap=148，表明第一个控制器有1个硬盘，第二个控制器有4个硬盘，第三个控制器有8个硬盘。

3、DiskIdxMap=这个参数值，按顺序从左到右每两位数为一个控制器的盘序数值，这两位数为16进制数，如果数值小于等于9或者大于10的，需要在数值前面加个0。比如DiskIdxMap=000105的意思是，第一个控制器的磁盘（引导盘）从序号1（00为序列1）开始往后排，第二个控制器的4个硬盘从序号2（01为序号2）开始往后排，第三个驱动器的8个硬盘从序号5（05为序号6）开始往后排。此时在群晖DSM7.X的系统里面看到磁盘序号就会从磁盘1（引导盘）到磁盘13。

4、这里有个小技巧，可以将控制器的盘序设置超过当前最大磁盘数量。例如我想把第1个控制器的引导盘隐藏，可以改成DiskIdxMap=0C0004，引导盘在群晖系统里面就看不到了，此时在群晖DSM7.X的系统里面看到磁盘序号就会从磁盘1到磁盘12（引导盘被隐藏了），同时还治愈了很多患有强迫症的童鞋！

5、SasIdxMap为SAS硬盘的个数，使用SATA硬盘的时候默认设置为0。

SiegFish, 2023-04-21:
arpl没有grub文件，你可以到引导里面的高级设置-查看所有sata端口和硬盘里查看你的硬盘是否被识别出了。
我遇到了类似的问题，但是我被屏蔽的是31和32号盘。aprl第二个控制器，就是直通sata控制器的位置应该是从31号开始的你可以再检查一下。

目前我已经换回了原版arpl，没有出现类似问题，暂时先不升级了。

wjz304, 2023-04-22:
少数的情况就用旧版吧。

closed 2023-04-25 by wjz304