RROrg/rr #6: Please try to ask questions on Github as much as possible.

wjz304, 2023-04-25:
Due to the high restrictions on email and IP in xpenology.com, I am unable to reply on xpenology.com. I am sorry.

raelix, 2023-04-29:
could you please help with the steps? I'm not really able to follow all the steps with translator :(

wjz304, 2023-04-29:
Do you need the steps of use?
you can see https://www.youtube.com/watch?v=mmwKCOiHGWA

raelix, 2023-04-29:
Thank you @wjz304 , the main issue I have is that the face recognition application doesn’t work, everything else looks working. I tried different build bus same behavior…is this expected?
Thanks a lot in advance!

wjz304, 2023-04-29:
The face recognition application is quite complex,  it is related to graphics cards, drivers, SN ... 
You need to find some tutorials now.
The instructions I found in China may not be easy for you to understand, I suggest you search on xpenology.com .

raelix, 2023-04-29:
thank you very much I found something about the Photos application but not about Surveillance face recognition. Do you think is it feasable without a GPU?

wjz304, 2023-04-29:
https://github.com/jinlife/Synology_Photos_Face_Patch

raelix, 2023-04-29:
Yes this is just for Synology Photos and it's for the older version (the current one is 1.4.0) but I'm searching for Surveillance face recognition not for Photos :( (sorry if I'm bothering you on this)...

wjz304, 2023-04-29:
You need to install the model of DVA xxxx, such as DVA3221
The DVA model seems to come with these features, but since I don't have the relevant hardware, I'm not very familiar with it

raelix, 2023-04-29:
last question :) so do you think I can use using the CPU with the DVA3221 without GPU? thanks again!!

wjz304, 2023-04-29:
I really don't know, I can help you ask other people during the day, that everyone is resting now

raelix, 2023-04-29:
oh sorry!! I didn't know (it almost night even here xD) ... Thank you! Super! The main thing I didn't get on using a specific model is if I should use tinycore-redpill or it's not needed

wjz304, 2023-04-30:
Sorry, I haven't received a response from anyone else,
Regarding Surveillance face, the results of tinycore-redpill and arpl are the same.

raelix, 2023-04-30:
Thank you very much for your help @wjz304 . 
do you know if there is a way of installing extra drivers?
I found that I can use the Surveillance face with dva1622 with just the CPU but I have a 10th gen (i7-10700k) and the driver are not recognized.

wjz304, 2023-04-30:
https://xpenology.com/forum/topic/59909-i915ko-backported-driver-for-intel-10th-gen-ds918-ver-701-up3

raelix, 2023-04-30:
I owe you a beer :) do you know who can I apply them to arpl? Thanks for your quick answer

wjz304, 2023-04-30:
[i915-8th.tar.gz](https://github.com/wjz304/arpl-i18n/files/11361134/i915-8th.tar.gz)
[i915-10th.tar.gz](https://github.com/wjz304/arpl-i18n/files/11361135/i915-10th.tar.gz)

raelix, 2023-04-30:
Still learning :( how can I use these files you provided? Where I should copy the content? Thanks thanks thanks

wjz304, 2023-04-30:
you can from "main menu -> Addons -> Download a external addon " add it. and install it

raelix, 2023-04-30:
Thank you @wjz304 ill try it in few hours and I’ll give you a feedback. My hardware is intel i7-10700K with integrated gpu

raelix, 2023-04-30:
Hi @wjz304  when I added it using the "download a external addon" but how to install it? Thanks :)

wjz304, 2023-04-30:
[image]
[image]
当你选择的型号和版本是该addon支持的时候，他将显示
[image]
params 不需要填写
[image]

raelix, 2023-05-01:
It is downloaded but doesn't appear in "select addons"...any suggestion?

closed 2023-05-15 by wjz304