RROrg/rr #10: qlogic bnx2x 10gbe  lan geminilake no work.

category-9431, 2023-04-26:
if nx2x-e2-7.12.30.0.fw file is in /lib/firmware/bnx2x location, it works.

wjz304, 2023-04-26:
thanks for the heads up.

wjz304, 2023-04-27:
I updated it, 
plz update modules in the update menu.

category-9431, 2023-05-01:
dsm 6 7 geminilake works. Thank you!

wjz304, 2023-05-02:
> dsm 6 7 geminilake works. Thank you! in case ds920+ 7.2 fresh install, firmware bnx2x-e2-7.12.30.0.fw is not located, After I copy it, works again. Can Loader put firmware in it?

每个版本的驱动依赖的fw不一样，人为处理全部平台的全部驱动的工作量很大。所以现在都是脚本在自动处理，
可能有错误我没有察觉，之后我会进行检查。

wjz304, 2023-05-02:
我检查了920 7.2 的依赖，貌似没发现问题，正常他会自动拷贝
[image]
[image]

category-9431, 2023-05-02:
OK! firmware file is right.
but After install, firmware files is not located in /lib64/firmware/bnx2x/*.fw.
I have to copy manually from firmware.tgz.

wjz304, 2023-05-02:
I will test it again

wjz304, 2023-05-05:
I have recompiled eudev and adjusted the service. If it still doesn't work well, manually copy it just for now,
Let's wait until the official version is launched,

wjz304, 2023-05-27:
plz try the new version.

category-9431, 2023-05-27:
OK really works on native ds920+. Thank you.

closed 2023-05-27 by wjz304