RROrg/rr #11: DSM 7.2 RC Release

mdstoll, 2023-04-26:
DSM 7.2 RC was recently released:
https://www.synology.com/en-us/support/download/DS1621xs+?version=7.2#system

7.2 build 64551 RC

Anyone tried updating already? Is it safe without updating the ARPL loader?

wjz304, 2023-04-26:
请更新到 23.4.4 后再升级 7.2RC

mdstoll, 2023-04-26:
Hmm. I downloaded the latest arpl from your repo and installed / configured it on the USB. I selected the 64551 config.

After booting, I see this 'migration' placeholder. I would very much like to not lose configuration but I guess that is too late?
https://imgur.com/jAPO8oG

wjz304, 2023-04-26:
如果你还没有 reinstall ， 我有一个想法，

1. 你把usb装置刷回 23.4.3， 他应该能正常的进入系统。
2. 在 DSM 系统内 手动升级 DSM 到 64551. 等它重启后会失败并自动进入到 arpl。
3. 此时关机，把usb装置刷到 23.4.4，在进入arpl，install 的 64551.
4. 启动。

试一下吧。

wjz304, 2023-04-26:
由于我的英语不好，我怕我说的散装英语你更加无法理解。抱歉。

mdstoll, 2023-04-26:
No worries, I'll just google translate your text ;)
I'll check that solution out. Will report back.

wjz304, 2023-04-26:
NVMe 的问题 是必然的，因为[nvme-cache.sh](https://raw.githubusercontent.com/PeterSuh-Q3/tcrp-addons/master/nvme-cache/src/nvme-cache.sh) 修改了系统文件，更新系统后，系统文件被还原了

wjz304, 2023-04-26:
由于驱动(modules)的问题比较棘手，暂时没有太多时间关注 插件(addons)，

mdstoll, 2023-04-26:
Thanks a lot for your help @wjz304 , has been working ever since I committed the changes.

closed 2023-05-19 by wjz304