RROrg/rr #7: update.zip的升级包怎么使用

y723157510, 2023-04-26:
update.zip的升级包怎么使用

wjz304, 2023-04-26:
update.zip 是 旧版本在线升级用的， 菜单-更新-更新arpl

closed 2023-04-26 by wjz304

dgulan, 2023-08-17:
我搜到了这个issue。想讲一下。

背景：ESXi虚拟机安装好了黑裙，正常使用中。

看到了有更新，于是想把【arpl-i18n-23.8.7.img.zip】 更新一下。
问题来了：怎么进入菜单？
本人小白，也不敢乱试。

我在issue、Instructions 里面，没找到进入菜单的方法。（自己白白的，也有可能写了，只是我没看懂）

最终，通过方法如下，更新了：
1、instructions 里关于“dsm下重启到arpl(免键盘) (Any version)“，通过ssh 链接黑裙，sudo -i以后，输入命令/usr/bin/arpl-reboot.sh "config"
2、黑裙会在2～3分钟内重新启动，然后根据提示进入 web，比如192.168.1.xx:7681 顺利进入菜单
3、选择更新，在线更新碰到问题，因为github 访问不了
4、下载update\addons\modules\rp-lkms 这4个包，分别上传，然后会更新。更新结束后，自动重启，顺利进入到DSM

DaDaDaBocai, 2023-08-18:
可是在线更新，变得好慢好慢。甚至升级不成功，从某一个版本开始。打开github是秒开。不知道为啥。

time2049, 2023-08-24:
> 

十分感谢，之前就很困惑这个问题，终于知道了解决方案。

dgulan, 2023-08-25:
其实我也是小白。

在懂的人眼中，这些都是很细微、常见的操作，熟悉到都忽略了讲出来。于我，却是需要反复折腾，试错。
把我的操作过程写出来，希望能帮到需要的人。