RROrg/rr #14: gnu grub页面依然无法自动倒计时并自动跳过

itechsen, 2023-04-27:
B365+i9-9900

itechsen, 2023-04-28:
两种都试过了，我看你有一个分支包是解决这个问题的。我试一下看看能不能解决这个问题

itechsen, 2023-04-28:
> 两种都试过了，我看你有一个分支包是解决这个问题的。我试一下看看能不能解决这个问题

grub-editenv: error: cannot open `/mnt/p1/grub/grubenv.new': No such file or directory.

wjz304, 2023-04-28:
那个是我临时修改的，启动会有问题，只看 grub 就好了，
虽然倒计时的问题没解决，但是我还是会投上去，等下一个版本试试吧，

如果不好使，尝试 删除 \p1\boot\grub\grub.cfg 里 前面的 insmod 的部分(1-12行)。

itechsen, 2023-04-29:
> 那个是我临时修改的，启动会有问题，只看 grub 就好了， 虽然倒计时的问题没解决，但是我还是会投上去，等下一个版本试试吧，
> 
> 如果不好使，尝试 删除 \p1\boot\grub\grub.cfg 里 前面的 insmod 的部分(1-12行)。

我还有一台机子 倒计时就正常，所以是不是还是bios设置里的某项配置参数导致的呢

ZeratullZHou, 2023-04-30:
我也是这个问题，之前星际蜗牛矿渣那台就没问题，最近换了b365M主板+I3-8100后就出这个问题了。感觉是不是BIOS的问题

wjz304, 2023-04-30:
有关系

EdikKing, 2023-05-01:
同遇到这个问题，主板是小厂的工控B365M主板

EdikKing, 2023-05-02:
问卖家要了个最新的bios，解决这个问题了，倒计时可以用了

lialonex1, 2023-05-02:
打开CSM，关闭快速启动试下呢？我用ARPL1.1beta2a遇到这个问题就这样解决了的。

lialonex1, 2023-05-02:
> 我也是这个问题，之前星际蜗牛矿渣那台就没问题，最近换了b365M主板+I3-8100后就出这个问题了。感觉是不是BIOS的问题

你的蜗牛用ARPL1.1b2a有没有问题？我的蜗牛和另外一个j3160的安装完成后，正常关机断电，再开机就启动时卡在正在启动…

ZeratullZHou, 2023-05-02:
> 问卖家要了个最新的bios，解决这个问题了，倒计时可以用了

你是用的雲星的主板吗，BIOS能发我一份不

iooy, 2023-05-23:
可以尝试修改BIOS启动项的倒计时，我也曾经为这个问题苦恼了两天，无意中修改BIOS中的倒计时就好了，随便改一个数字保存重启测试下。

doandat943, 2023-07-29:
@itechsen is there any new information, in the old versions it worked like a charm, but since the last two versions I have had the same situation as you

wjz304, 2023-09-21:
No further feedback, this issue has been closed. If there are any issues, please create a new issue

closed 2023-09-21 by wjz304

wjz304, 2023-09-22:
总结:
1. 更新 BIOS.
2. 修改 BIOS 启动项的倒计时.
3. 修改 BIOS 中找 "super io" , "serial port" 相关的选项》
4. 修改 set timeout="5" 为 0 （/mnt/p1/boot/grub/grub.cfg）