RROrg/rr #15: PVE安装的arpl-i18n，建立引导之后arpl无法获得ip地址

theseato, 2023-04-27:
[image]

如图，建立引导之后，无法获取IP地址。

theseato, 2023-04-27:
由此导致的问题是无论用什么办法都找不到自己的nas。

theseato, 2023-04-28:
问题好像是出在不能设置多网卡。目前我的情况就是，只设置mac1，能正常运作。如果设置了mac2、mac3……就不能正常运行了。（如果不能设置多网卡，就面临不能洗白）

跟pve虚拟机的网卡模式选定似乎没有关系。

wjz304, 2023-04-28:
能说明一下 实际网卡数和 自定义的 mac数吗？

wjz304, 2023-04-28:
现在这块的情况确实比较复杂，我确实有很多情况欠考虑了

wjz304, 2023-04-28:
另外 这个
[image]
只是一个提示性信息，我写错了，显示并不准确，请忽略他，下个版本将修复。

wjz304, 2023-04-29:
try the new version

theseato, 2023-04-30:
新版本解决了设置多个mac情况下arpl无法获取的ip的情况，但是……查找不到系统，无论是多mac设置还是单mac设置，都无法找到群晖。

wjz304, 2023-04-30:
> 新版本解决了设置多个mac情况下arpl无法获取的ip的情况，但是……查找不到系统，无论是多mac设置还是单mac设置，都无法找到群晖。
PVE 加串口，看日志

theseato, 2023-04-30:
> > 新版本解决了设置多个mac情况下arpl无法获取的ip的情况，但是……查找不到系统，无论是多mac设置还是单mac设置，都无法找到群晖。
> 
> PVE 加串口，看日志

现在这情况很尴尬。我退回了之前用的那个v23.4.4，但升级了addons、LKMs、modules等，升级了之后用v23.4.4建立的引导也无法查找到群晖系统。

重新下载一个v23.4.4来重新做引导，“Build the loader”这步又会报错（“Hash of pat not match,……”）。
[image]

目前就卡在这里了，高低版本都用不了。

wjz304, 2023-04-30:
是的，5本来不在发布计划之内，但是官方 更新了pat。

wjz304, 2023-04-30:
url: "https://global.synologydownload.com/download/DSM/release/7.2/64551/DSM_DS3617xs_64551.pat"
      hash: "17785c8dab0cc2faf4c0ac3d6ddca3c38c782fc57586004182ae09e443410db4"
      ramdisk-hash: "212e03d267917774ebd3fbb710802cda6e2b674541d89df200a09ff5b28af7bb"
      zimage-hash: "16176d29125ee784ab8bfb1201f0269127ca70464f86022ad96e0023b2c95308"
      md5-hash: "3ba13be674c7e51114ffb91484e60259"

wjz304, 2023-04-30:
你可以自己改它 /opt/arpl/model-configs/DS3617xs.yml

theseato, 2023-04-30:
> 是的，5本来不在发布计划之内，但是官方 更新了pat。
用V23.4.4重新做了引导，还是查找不到群晖。思来想去，难道是这个最新版本pat，跟群晖系统上的之前的那个pat不比配，导致无法找到系统？？

wjz304, 2023-04-30:
不， 我也很奇怪，PVE 突然就会报 eudev 的错误，然而就 eudev 这个addons 没有修改， DTB的型号貌似不会，Q35下 3622 有50%的概率，...变化只有 我昨天更新了pve

wjz304, 2023-05-01:
try [23.5.0](https://github.com/wjz304/arpl-i18n/releases/tag/23.5.0)

theseato, 2023-05-01:
> try [23.5.0](https://github.com/wjz304/arpl-i18n/releases/tag/23.5.0)
依旧无法正确的找到群晖系统。
[image]

好消息是……“https://finds.synology.com/”能工作了，坏消息是它找出来的设备相关信息全部错误。

[image]

wjz304, 2023-05-01:
finds.synology.com 会受到缓存干扰，准确，以紫色的为准（但是 也会出现 ip变动的情况）

theseato, 2023-05-02:
> finds.synology.com 会受到缓存干扰，准确，以紫色的为准（但是 也会出现 ip变动的情况）
进入群晖系统了。因为我设置的是手动ip，之前更新后就直接从设置手动ip就能进入系统，这次要先进入arpl给出的地址，点击一下还原，然后才能正常切换到设置的手动ip那里。
V23.5.0，成功。

closed 2023-05-02 by theseato

zhaoleijack, 2023-07-02:
[image]
群晖里面重启后失联。但是关闭虚拟机，再启动虚拟机后能正常登录.

wjz304, 2023-07-02:
> [image] 群晖里面重启后失联。但是关闭虚拟机，再启动虚拟机后能正常登录.

https://qun.qq.com/qqweb/qunpro/share?_wv=3&_wwv=128&appChannel=share&inviteCode=20jx8dPsU2z&contentID=1m4NKs&businessType=2&from=181174&shareSource=5&biz=ka