RROrg/rr #16: 24.4.4 not woring

green1052, 2023-04-27:
i installed DS3622+ with console addon

when i boot the dsm 
stuck "BOOTING..." PAGE

model: 3622+
version: 54551
NIC: BRCM 10GbE 2P 57810S-t

green1052, 2023-04-27:
and i think doesn't wokring my network card (`BRCM 10GbE 2P 57810S-t`)
i set manually mac1 but i cant access web ui

wjz304, 2023-04-27:
I updated it, 
plz update modules in the update menu.

wjz304, 2023-04-27:
If the Mac is customized, there may be a prompt for obtaining an IP error on the boot page. Please ignore it and I will fix it in the next version.

green1052, 2023-04-27:
sadly not working

green1052, 2023-04-28:
arpl 2.4.4.2 was worked for me

green1052, 2023-04-28:
[#15](https://github.com/wjz304/arpl-i18n/issues/15#issuecomment-1526872043) same issue

green1052, 2023-04-28:
i edit manually [boot.sh](https://github.com/wjz304/arpl-i18n/blob/main/files/board/arpl/overlayfs/opt/arpl/boot.sh)
finally i fixed!