RROrg/rr #19: arpl-i18n-23.4.5.img问题

host527, 2023-04-30:
（1）配置完成后启动失败，需要手动关机在开机才能进入。
（2）引导时显示器显示乱码，远程web界面正常。

wjz304, 2023-04-30:
问题1 应该和 #18 是一个问题

wjz304, 2023-05-01:
try [23.5.0](https://github.com/wjz304/arpl-i18n/releases/tag/23.5.0)

host527, 2023-05-01:
刷了arpl-i18n-23.5.0.img，重启完还是死循环，刷新网页就是10分钟倒计时。主机上也没有显示，显示器显示无信号

CarlosLucky, 2023-05-03:
> 刷了arpl-i18n-23.5.0.img，重启完还是死循环，刷新网页就是10分钟倒计时。主机上也没有显示，显示器显示无信号

你的问题和我一样，请问你是不是多网卡，板载一个1G （如8111H）  pcie一个网卡（8125），arpl识别出两个网卡mac1为pcie，mac2为板载1G ，进入DSM识别出mac1 为原生板载 mac2 为pcie ，总结就是arpl读的网卡和DSM读的刚好顺序反了，导致我的路由器启动aprl时候第一次分配到的ip和DSM分配的冲突因为mac调换了，在arpl的显示页面显示mac1 ..... 错误 mac2 ......错误 之后搜不到任何设备，这个时候只需要重启一下主机路由器重新分配ip对应到网口，你才能连到设备。

wjz304, 2023-06-12:
> 你的问题和我一样，请问你是不是多网卡，板载一个1G （如8111H） pcie一个网卡（8125），arpl识别出两个网卡mac1为pcie，mac2为板载1G ，进入DSM识别出mac1 为原生板载 mac2 为pcie ，总结就是arpl读的网卡和DSM读的刚好顺序反了，导致我的路由器启动aprl时候第一次分配到的ip和DSM分配的冲突因为mac调换了，在arpl的显示页面显示mac1 [...]

Realtek 网卡看最新的测试版，有问题反馈，
这个issue没有参考价值先关闭了。

closed 2023-06-12 by wjz304