RROrg/rr #20: about kernel 4.4.302+ invalid module format, unknown symbol issue

PeterSuh-Q3, 2023-05-01:
Hello.

I tried using apollolake-4.4.302.tgz file in my M SHELL for TCRP in the modules.zip you compiled.

M SHELL for TCRP DSM 7.1 used arpl-modules directly.
Currently, I am also using arpl-modules that I compiled myself after forking.

However, apollolake-4.4.302.tgz made with the 7.2 compiler prepared by fabio had an unknown symbol and  invalid module format issue.

https://github.com/pocopico/redpill-modules/issues/1

That issue I confirmed with pocopico a few weeks ago.

I tried using your apollolake-4.4.302.tgz on TCRP DSM 7.2 RC and DS1019+ and the junior log gave the same " invalid module format, unknown symbol".

In your arpl all modules work without this "unknown symbol" message?

wjz304, 2023-05-01:
I have made some replies on https://github.com/pocopico/redpill-modules/issues/1

and，
[image]

There are also unknown symbols in my package, but they have little impact on startup.
You should remove the currently problematic drivers or be able to continue running if loading fails.

wjz304, 2023-05-01:
They also run well on redpill_load,
Currently, Same of drivers and lkms for both my [Redpill_CustomiBuild](https://github.com/wjz304/Redpill_CustomBuild) and arpl-i8n.
https://github.com/wjz304/rp-ext

wjz304, 2023-05-01:
Also, if you know the management personnel of xpenology.com, I hope xpenology.com can lift the restrictions on my email, thanks! :)

closed 2023-05-03 by PeterSuh-Q3