RROrg/rr #21: EXSI7.0环境下，通过HBA330直通硬盘，安装DS3620XS+一点问题没有，就是安装DS918+死活不认盘

xinjing-cloud, 2023-05-01:
EXSI7.0环境下，通过HBA330（DELL) 直通硬盘，也尝试用H310(DELL 直通卡）安装DS3620XS+一点问题没有，所有硬盘都能正常识别。但是安装DS918+死活不认盘，总是提示找不到硬盘。是因为DS918+没有阵列卡的驱动吗？

wjz304, 2023-05-01:
支持的驱动有差异，系统内的判断也有差异，暂时没太多时间解决单点问题。

jiahuasiyu, 2023-05-02:
不要直通卡，添加硬盘（HBA330连接的盘）即可！

AuxXxilium, 2023-05-07:
DS918+ doesn't support SAS/SCSI or Raid Cards.

wjz304, 2023-05-07:
> DS918+ doesn't support SAS/SCSI or Raid Cards.
@AuxXxilium 
你懂中文了啊？
[image]

AuxXxilium, 2023-05-07:
> > DS918+ doesn't support SAS/SCSI or Raid Cards.
> 
> @AuxXxilium 你懂中文了啊？ [image]

no, but google translater ;)

wjz304, 2023-05-07:
> > > DS918+ doesn't support SAS/SCSI or Raid Cards.
> > 
> > 
> > @AuxXxilium 你懂中文了啊？ [image]
> 
> no, but google translater ;)

Do I speak English with too many syntax error?

wjz304, 2023-07-03:
长时间未回复，如新版有问题，请重新创建issue

closed 2023-07-03 by wjz304