RROrg/rr #26: DSM 7.2 ACPID module does not work

OrpheeGT, 2023-05-02:
Hello,

Requesting shutdown from Proxmox GUI does not work.

The request is not triggered.

From Proxmox :
TASK ERROR: VM quit/powerdown failed - got timeout

wjz304, 2023-05-02:
是的，目前没有 7.2 的 acpid

Odroid13, 2023-05-26:
> 是的，目前没有 7.2 的 acpid，当前是7.1的.

7.2 RC It's Work 

https://github.com/wjz304/arpl-i18n/assets/97029183/a41b13e2-93ec-4bbd-ab18-61c39be3b06f

OrpheeGT, 2023-05-26:
Hello,

I agree it has been fixed in the meantime.

closed 2023-05-26 by OrpheeGT

Odroid13, 2023-05-26:
> Hello,
> 
> I agree it has been fixed in the meantime.

At home works on 7.2 RC version and not on Final version.
How did you solve the problem in version 7.2?

Thanks

OrpheeGT, 2023-05-26:
I can't tell, I'm running ARPL 23.5.8 on Proxmox VM with 7.2 final and it just works.

Odroid13, 2023-05-26:
> I can't tell, I'm running ARPL 23.5.8 on Proxmox VM with 7.2 final and it just works.

Can you give me a screenshot so that I can see the details of your VM?

OrpheeGT, 2023-05-26:
Nothing anusual, standard QC35 UEFI machine.

mrfy, 2023-05-28:
> > 是的，目前没有 7.2 的 acpid，当前是7.1的.
> 
> 7.2 RC It's Work
> 
>  241063164-5361f724-15dd-4037-9ae5-7fed066a7ce4.mp4

Hi @Odroid13 , is this exactly how i can update my baremetal DSM from 7.1.5 to 7.2? :) Thanks

wjz304, 2023-05-28:
> > > 是的，目前没有 7.2 的 acpid，当前是7.1的.
> > 
> > 
> > 7.2 RC It's Work
> > 241063164-5361f724-15dd-4037-9ae5-7fed066a7ce4.mp4
> 
> Hi @Odroid13 , is this exactly how i can update my baremetal DSM from 7.1.5 to 7.2? :) Thanks

只能说明再某些情况下，他们兼容