RROrg/rr #31: Network lost after DSM boot

sashakozlov86, 2023-05-07:
Hello, friends! On a baremetal laptop HP x360 Pavilion ba103ur (connected via USB3.0-LAN Adatep D-Link DUB-1312, also tried TP-link) the problem - after a certain period of time (after determining the IP address and DSM kernel boot), the device disappears in the router.

If i boot in config bootloader mode, then the network does not disappear.  This happens after loading DSM.  I've already tried disabling the "cdc ncm" module, but that doesn't help.  Thanks for any help!

AuxXxilium, 2023-05-07:
No module/driver for ax88179 chip

sashakozlov86, 2023-05-07:
Do you mean that module/driver for ax88179 chip needs to be installed?  To be honest, I chose "select all modules" and simply excluded "cdc ncm" from this entire list of modules, although there is no difference with and without this module (cdc ncm).  Disabling "cdc ncm" used to work in older versions of arpl, but not now.  Regarding the module / driver for ax88179 chip - do I need to install it separately (it is not in the list of all modules)?  Thanks for your reply!

wjz304, 2023-05-07:
你要安装什么型号？

sashakozlov86, 2023-05-07:
DVA1622

wjz304, 2023-05-07:
[modules-test.zip](https://github.com/wjz304/arpl-i18n/files/11415888/modules-test.zip)

把他解压后的文件放到 /mnt/p3/modules 下，然后重新编译

wjz304, 2023-05-07:
刚刚加的编译出来，如果它好使， 下一个版本将发布

AuxXxilium, 2023-05-07:
@wjz304 it is possible that usbnet.ko is working too for this chip.

sashakozlov86, 2023-05-07:
> 刚刚加的编译出来，如果它好使， 下一个版本将发布

Dear friend, thank you very much!  That is, if everything works, then in the next version we can expect a fix for this bug?  I honestly do not quite understand how you can compile.  Thank you!

wjz304, 2023-05-07:
> @wjz304 it is possible that usbnet.ko is working too for this chip.

OK, 一会儿我看一下 usbnet.c

wjz304, 2023-05-07:
> > 刚刚加的编译出来，如果它好使， 下一个版本将发布
> 
> Dear friend, thank you very much! That is, if everything works, then in the next version we can expect a fix for this bug? I honestly do not quite understand how you can compile. Thank you!

Yes, of course you need to test it first to see if it works

sashakozlov86, 2023-05-07:
> 

Okay.  And what file should be downloaded and how to compile?

wjz304, 2023-05-07:
> [modules-test.zip](https://github.com/wjz304/arpl-i18n/files/11415888/modules-test.zip)
> 
> 把他解压后的文件放到 /mnt/p3/modules 下，然后重新编译

This attachment
Download it and unzip it, and place the three files inside under/mnt/p3/modules in arpl to replace the original files

sashakozlov86, 2023-05-07:
Thank you!  It turns out that it is better to unpack the img image using an archiver, replace the file there in the specified folder and archive it back into img?

wjz304, 2023-05-07:
I don't think I can explain this matter clearly in English.

Just a moment, I'll give you a beta version.

wjz304, 2023-05-07:
https://github.com/wjz304/arpl-i18n/releases/download/23.5.1/arpl-test-ax88179.zip

sashakozlov86, 2023-05-07:
Okay, thank you very much dear friend!  I'll try to overwrite the flash drive tomorrow evening and check.  Then I will write to you.  Should I just select the ax88179 module?

wjz304, 2023-05-07:
> Do you mean that module/driver for ax88179 chip needs to be installed? To be honest, I chose "select all modules" and simply excluded "cdc ncm" from this entire list of modules, although there is no [...]

Following your original approach, give it a try.

sashakozlov86, 2023-05-07:
Okay friend!  Thank you very much!  I'll definitely try it tomorrow night and let you know the result.  Have a nice day!

wjz304, 2023-05-07:
https://github.com/wjz304/arpl-i18n/releases/download/23.5.1/arpl-test-ax88179-2.zip
我检查了一下，它有 usbnet 的依赖，我重新编译了一下，请用这个测试

sashakozlov86, 2023-05-08:
Unfortunately friend, it didn't work.  After installing DSM and rebooting the computer, it disappears from the network when loading DSM

AuxXxilium, 2023-05-08:
@wjz304 you modules build is crashed. ;)

sashakozlov86, 2023-05-08:
I also tried with USB3.0-LAN adapter TP-link UE300.  It also does not work with it, the device disappears from the network.

wjz304, 2023-05-08:
如果 你知道如何进入shell，可以先测试 ko 是否好使。

sashakozlov86, 2023-05-08:
Unfortunately I don't know how to do it.  When configuring the bootloader, I write "menu.sh" and then I do the rest of the actions through the menu.

AuxXxilium, 2023-05-08:
usb:v0B95p1790*
 ID_MODEL_FROM_DATABASE=AX88179 Gigabit Ethernet

from: 20-usb-vendor-model.hwdb

wjz304, 2023-05-08:
> usb:v0B95p1790* ID_MODEL_FROM_DATABASE=AX88179 Gigabit Ethernet
> 
> from: 20-usb-vendor-model.hwdb

There are still some issues with my modified eudev

wjz304, 2023-05-08:
@sashakozlov86  Just a moment, I'll give you a restored version of eudev

AuxXxilium, 2023-05-08:
and:

usb:v2001p4A00*
 ID_MODEL_FROM_DATABASE=DUB-1312 Gigabit Ethernet Adapter for this specific version

sashakozlov86, 2023-05-08:
My TP-Link UE300/UE330 USB to Ethernet adapter based on Realtek RTL8153

wjz304, 2023-05-08:
7.2 or 7.1 ？

sashakozlov86, 2023-05-08:
7.2

AuxXxilium, 2023-05-08:
@wjz304 the rtl8153 is in eudev set by "50-usb-realtek-net.rules"

wjz304, 2023-05-08:
> @wjz304 you modules build is crashed. ;)

emm.... I just noticed..

sashakozlov86, 2023-05-08:
> @wjz304 the rtl8153 is in eudev set by "50-usb-realtek-net.rules"

That's just the point, I select all modules, that is, in fact, all drivers should be installed, including on my adapter, but for some reason the network still disappears when DSM is loaded.  Up to this point, the adapter is determined in the bootloader and the IP address is also determined, but then the device disappears.

wjz304, 2023-05-08:
https://github.com/wjz304/arpl-i18n/releases/download/23.5.1/arpl-test-ax88179-3.zip

wjz304, 2023-05-08:
https://github.com/wjz304/arpl-i18n/issues/10

Yes, according to this issue, there is a problem with the 7.2 driver
I have also tried some other drivers that can be manually loaded, but they do not automatically load.
The rules for eudev in 7.1 and 7.2 are consistent, and it is currently unknown where the problem lies

wjz304, 2023-05-08:
So I tried to recompile eudev, but I feel like it's not working well, 
Although it is possible to attach an action and copy it again to drive it to the system for loading, this is not a good solution.

wjz304, 2023-05-08:
So these are two questions
1. Is the driver available
2. Does the driver automatically load

AuxXxilium, 2023-05-08:
good question, need console to find out

wjz304, 2023-05-08:
But I don't have these hardware, so I can only try to solve problem 2 first and then rely on others to give me the answer to problem 1, 
[image]

sashakozlov86, 2023-05-08:
> https://github.com/wjz304/arpl-i18n/releases/download/23.5.1/arpl-test-ax88179-3.zip

Unfortunately it didn't help.

wjz304, 2023-05-08:
Let's give up for now and wait for further information from me,

Of course, you can also try manually loading it in the system and providing feedback, (But if you don't know how to operate it, you don't have to try, just wait for me.)
```
insmod mii.ko
insmod usbnet.ko
insmod ax88179_178a.ko
```

wjz304, 2023-06-12:
试一下 ax88179_178a ？

wjz304, 2023-06-18:
modules 23.6.9

closed 2023-06-18 by wjz304