RROrg/rr #32: 从最开的版本到现在最新的23.5.3，装蜗牛星际D的机器就没有成功过。

xinjing-cloud, 2023-05-08:
家里有一个蜗牛D的机器，一直用来放照片的。用JUN大的7.0.1的引导，一直到现在也随着群晖的升级到7.1.1 U5都是可以正常引导的，唯一的遗憾就是经常会出现无法关机的情况。用您的所有版本，要么第一次启动的时候找不到硬盘（已经设置了DiskIdxMap=0001，SATA1号口装了一个120GSSD SATA，其他扩展的4个口装了4个2TB机械），要么就是找到硬盘后装好系统也能进入桌面一次，仅仅只有一次能进入桌面，再重启后网卡灯始终亮不起来，不知道哪个环节出了问题，请赐教！

wjz304, 2023-05-08:
没成功过就用其他的么，何必如此执着。

xinjing-cloud, 2023-05-09:
唯有执着方能成功

wjz304, 2023-05-09:
说实话，虽然你说了很多，但是没有一点是对问题有帮助的，我要和你沟通很久才能获取到一些有价值的信息，
然而大概率我也解决不了。

le45tx1m, 2023-05-12:
@xinjing-cloud 没有具体信息不好判断，但老实说看到DiskIdxMap=0001大概率就错了，arpl的sata驱动由原作者fbelavenuto进行过修改，一般不需要手动配置，编译的时候先去查看高级设置里认出了几块盘，或者改型号为DS920+，这个sata驱动模块是官方重写过的。[https://www.bilibili.com/video/BV1jk4y1Y7B7](url)

sybing01, 2023-05-23:
我也是蜗牛d的，没出现问题，你这些设置确定要设置？

wjz304, 2023-06-22:
无环境信息，无法分析，本issue暂时关闭.

closed 2023-06-22 by wjz304

wjz304, 2023-09-16:
https://github.com/wjz304/arpl-i18n/issues/219