RROrg/rr #35: 7.1.1-42962 U1 or U5 DSM latest version installation issue

PeterSuh-Q3, 2023-05-09:
It looks like you used the 7.1.1-42961-1 config.json I built.

However, because the Small FIX version has changed
If you proceed with 7.1.1-42962 during the DSM installation process,
It doesn't point to the 7.1.1 latest version of that version, it points to the 7.2 RC version.

like below
Adjustment of rss.json and rss.xml is required.

https://github.com/PeterSuh-Q3/redpill-load/commit/1270c0379970a7313de540d9e07fdbe4f7416649
https://github.com/PeterSuh-Q3/redpill-load/commit/85c4a630cc80a77fc7f580934c6e6e49b85baef0

I already separated 7.0.1 / 7.1.1 / 7.2.0 into folders like below.
The result of testing in this way
It has been confirmed that DS923+ and DS918+ operate normally.

The peculiarity is that DS923+ should point to U5 information, but it works together without problems with GM's setting.

Below is the GM buildphase information related to .VERSION U5 / U1 of the two models.
Please note.

./VERSION
majorversion="7"
minorversion="1"
major="7"
minor="1"
micro="1"
productversion="7.1.1"
buildphase="GM"
buildnumber="42962"
smallfixnumber="5"
nano="5"
base="42962"
packing="NM-nano"
packing_id="22"
derived="NM-nano"
derived_id="22"
builddate="2023/04/08"
buildtime="03:35:17"
unique="synology_r1000_923+"
extractsize=850836
partitionversionlimit=0
syno_supported_hwrevision="r0"
indexdbextractsize=1408
synohdpackimgextractsize=23508
packages=FileStation,SynoFinder,USBCopy,HighAvailability,OAuthService,SMBService,ScsiTarget,ActiveInsight,DhcpServer,SecureSignIn,HybridShare,Python2,BackupRestoreManager
packagerootextractsize=260500
packagevolumeextractsize=0

./VERSION
majorversion="7"
minorversion="1"
major="7"
minor="1"
micro="1"
productversion="7.1.1"
buildphase="GM"
buildnumber="42962"
smallfixnumber="0"
nano="0"
base="42962"
builddate="2022/09/22"
buildtime="18:00:57"
unique="synology_apollolake_918+"
extractsize=840820
partitionversionlimit=0
syno_supported_hwrevision="r0"
indexdbextractsize=1412
synohdpackimgextractsize=23508
packages=FileStation,SynoFinder,USBCopy,HighAvailability,OAuthService,SMBService,ScsiTarget,ActiveInsight,DhcpServer,SecureSignIn,HybridShare,Python2,BackupRestoreManager
packagerootextractsize=260520
packagevolumeextractsize=0

wjz304, 2023-05-09:
I know there's this problem,
But the SSR link is written to synoinfoconf,
If this address is modified, it will invalidate the address of the person who previously installed it, or cause difficulties for upgrading to larger versions.
There is currently no perfect solution,

wjz304, 2023-05-09:
Actually, I should have temporarily deleted the SSR for 7.2, but most people also came for 7.2, so I didn't do that

wjz304, 2023-05-09:
Due to my poor English, I may not be able to understand your meaning correctly. I will test whether using “RC/GM” can change them.

PeterSuh-Q3, 2023-05-09:
如果多个项目（版本）在统一的 rss.xml 中失败，尝试像下面的 M SHELL 那样将它们分开。

[image]

wjz304, 2023-05-10:
I understand what you mean, but this will bring several problems

1. Incompatible with someone who has already been installed.
2. Unable to obtain updates for larger versions. Or it can modify the SSR's pointing address. But most people don't know how to operate it

Is there a better solution?

PeterSuh-Q3, 2023-05-10:
ARPL and TCRP may be in a slightly different situation.
TCRP is defined as follows in config.json of redpill-load.
In DSM, small update through the control panel is blocked.

By any chance, is ARPL open to this part?

     "rss_server": "https://raw.githubusercontent.com/PeterSuh-Q3/redpill-load/master/rss/7.1.1/rss.xml",
     "rss_server_ssl": "https://raw.githubusercontent.com/PeterSuh-Q3/redpill-load/master/rss/7.1.1/rss.xml",
     "rss_server_v2": "https://raw.githubusercontent.com/PeterSuh-Q3/redpill-load/master/rss/7.1.1/rss.json",
     "small_info_path": "https://example.com/null",
     "security_version_server": "https://example.com/smallupdate"

Because of these settings, TCRP allows only manual updates for small updates.

If you want to move from 7.1 to 7.2 you will need to rebuild the loader for 7.2.
Is ARPL capable of auto-updating to 7.2 DSM with only 7.1 loader?

[image]

wjz304, 2023-05-10:
[image]

Actually, ARPL allows users to  manually set rss_server, rss_server_ssl, rss_server_v2 in "menu", it can be modified to the address you are satisfied with, but it can only be modified before compilation.
users can directly use your RSS address.

wjz304, 2023-05-10:
In addition, because the parameters of each model in ARPL's current logic are shared, modifying it requires splitting, which will also increase maintenance times.

wjz304, 2023-05-10:
I think we can solve this problem by using local files
 https://github.com/wjz304/arpl-i18n/commit/cd20756771e64a26478d95aef864c61cb37a5b44

wjz304, 2023-05-11:
https://github.com/wjz304/arpl-i18n/commit/f14591ea972c321c5f7fdb5cb54320ce06fa26bd

closed 2023-05-14 by wjz304