RROrg/rr #36: 超过26个硬盘，出现Loader disk not found 错误

xuwenhong2023, 2023-05-11:
我在使用最新的23.5.3和23.5.5引导时，当通过直通卡2308或者3108连接硬盘时出现异常表现在：当连接的硬盘数超过26个（包含启动盘），在引导出现 Autmated RedPill loader，arpl login：root（automatic login） 时，出现Loader disk not found！然后整个引导停下来，我检查了一下，就是如果挂载的硬盘超过 /dev/sdz，既/dev/sdaa、/dev/sdab。。。 后就会出现这个异常就是超过26个盘符后就会出现，同时发现超过26个硬盘就没办法挂载启动盘的3个分区到/mnt/p1、/mnt/p2、/mnt/p3下，所以找不引导盘，所以我认为这是一个逻辑上的bug，可能死没考虑到硬盘超过26个后，硬盘挂载的名字从sdz三位数变成sdaa四位数。我不知道怎么修改，望可以更新
[image]

wjz304, 2023-05-11:
超过26个，syno 也不支持。 arpl改了 后面也会有问题

xuwenhong2023, 2023-05-12:
[image]

wjz304, 2023-05-12:
blkid

xuwenhong2023, 2023-05-14:
我分別用了 blkid、lsblk、和lspci 分別輸出結果如下，U盤的lable=arpl3卷標也是正確的,小於26個硬盤時/Dev/sdz sdz為3個字符，我估計是硬盤掛載為/Dev/sdaa的時候sdaa為4個字符，估計在正則表達式的時候沒有考慮到4個字符的bug
[blkid.txt](https://github.com/wjz304/arpl-i18n/files/11470882/blkid.txt)
[lsblk.txt](https://github.com/wjz304/arpl-i18n/files/11470884/lsblk.txt)
[lspci.txt](https://github.com/wjz304/arpl-i18n/files/11470885/lspci.txt)

wjz304, 2023-05-14:
https://github.com/wjz304/arpl-i18n/actions/runs/4970040229
试下这个，

xuwenhong2023, 2023-05-14:
[image]
漂亮，進去了，試一下看看

xuwenhong2023, 2023-05-14:
另外一個issue就是你主頁的捐贈碼，我用微信或者支付寶掃碼了，怎麼沒辦法識別

wjz304, 2023-05-14:
是微信，可以的吧

closed 2023-05-14 by wjz304

xuwenhong2023, 2023-05-14:
OK了,捐獻了100喝咖啡！