RROrg/rr #38: 蜗牛星际安装ds 920+

hou6807628, 2023-05-13:
Model: DS920+
Build: 42962
*** Detected 1 network cards, but only 0 MACs were customized, the rest will use the original MACs. ***

可以检测到ip，但是连接不上

qinger998, 2023-05-14:
那就对了

hou6807628, 2023-05-14:
why?不适合蜗牛星际？发自我的手机-------- 原始邮件 --------发件人： qinger998 ***@***.***>日期： 2023年5月14日周日 18:26收件人： wjz304/arpl-i18n ***@***.***>抄送： hou6807628 ***@***.***>, Author ***@***.***>主    题： Re: [wjz304/arpl-i18n] 蜗牛星际安装ds 920+ (Issue #38)
那就对了

qinger998, 2023-05-14:
看看网卡驱动吧

qinger998, 2023-05-14:
换918看看吧

xinjing-cloud, 2023-05-15:
网卡驱动的问题，我基本上测试了所有版本都是一样失联。等后续的版本看看能不能修复这个问题。

sybing01, 2023-05-23:
我是d的安装没问题

hou6807628, 2023-05-23:
用了英文版的，可以连接上，不过有时候机子会重启

Halooooooo, 2023-05-27:
> 用了英文版的，可以连接上，不过有时候机子会重启

那个版本啊老铁

hou6807628, 2023-05-27:
> > 用了英文版的，可以连接上，不过有时候机子会重启
> 
> 那个版本啊老铁

就是原版的英文版

closed 2023-06-06 by hou6807628

wjz304, 2023-09-16:
https://github.com/wjz304/arpl-i18n/issues/219