RROrg/rr #41: BCM5720 Module tg3 eth0 ERROR booting Loader

iTzSnicholls, 2023-05-16:
when booting from USB to configure the Loader eth0 works fine and get IP and i can configure hte loader. However when going to boot from the loader it failes to get MAC Addresses and then when trying to find IP's on eth0 and eth1 it fails with ERROR and then the screen locks up as normal to boot DSM.

HP Microserver G10, DSM 7.1.1-42962Update5

qinger998, 2023-05-17:
Mail me the equipment, and I'll see why.

wjz304, 2023-05-17:
Find IP's ERROR is just an informational message,
It is not necessarily that the DSM cannot obtain the IP, 
Recommended to check whether there is an assigned IP under the route.

iTzSnicholls, 2023-05-17:
Cheers will try

I reverted back to the ArpL v1.0beta2 in the end to recover my XPE data

wjz304, 2023-05-17:
There is also a situation where changing the location of the PCI results in a change in the ETH order, resulting in an error in the MAC address

wjz304, 2023-06-09:
重编了 tg3 驱动，方便的时候请测试.

wjz304, 2023-06-18:
长时间未得到回复，本issue暂时关闭。

closed 2023-06-18 by wjz304