RROrg/rr #42: intel i217-v集成网卡总是丢失连接

exeron21, 2023-05-17:
试过原版的arpl和arpl-i18n, 试过好多版本的引导，总是会丢失连接，具体现象有：
编译成功后网卡灯灭，Ping不通，群晖助手找不到，当然也无法访问
编译成功后网卡灯亮，Ping得通，但群晖助手找不到，5000端口无法访问
重启后获取不到Ip，eth0 error ，访问不了
重启后能获取到Ip，但网卡灯灭，Ping不通，群晖助手找不到，当然也无法访问

网卡是intel i217-v
[image]

wjz304, 2023-05-17:
ok，我看一下 i217-v 的驱动

exeron21, 2023-05-17:
我上哪去弄日志？
我自己查了一下,i217-v这个网卡应该是e1000系列的，驱动应该是igb.ko，我看到Models目录下是有这个驱动的，不知道怎么回事

exeron21, 2023-05-17:
> ok，我看一下 i217-v 的驱动

谢谢，这是我在Intel驱动包里扒出来的一段话： （intel在linux下的网卡驱动包是一个大包，700多M，对我来说有点难搞）
This release includes Linux* Base Drivers for Intel(R) Ethernet Network
Connections.

- igb driver supports all 82575-, 82576-, 82580-, I350-, I210-, I211- and
  I354-based gigabit network connections.
    - Gigabit devices based on the Intel(R) Ethernet Controller X722 are
      supported by the i40e driver.

驱动包里部分内容：
[image]

驱动包里翻出来的：
[readmefirst.txt](https://github.com/wjz304/arpl-i18n/files/11498573/readmefirst.txt)

Intel官网驱动下载地址：

qinger998, 2023-05-17:
没有这个型号的驱动

wjz304, 2023-05-17:
好比  21年的驱动也不能兼容 22年的网卡，要升级驱动的啦... 除非 新型号的 id 啥的 都不变，只是 name变

wjz304, 2023-05-17:
e1000e,
```shell
curl -kLO https://raw.githubusercontent.com/wjz304/rp-ext/main/modulesplatforms/apollolake-4.4.302/e1000e.ko
insmod e1000e.ko
```
看看报啥错

exeron21, 2023-05-17:
> e1000e,
> 
> ```shell
> curl -kLO https://raw.githubusercontent.com/wjz304/rp-ext/main/modulesplatforms/apollolake-4.4.302/e1000e.ko
> insmod e1000e.ko
> ```
> 
> 看看报啥错

现在唯一的U盘已经换了别的引导在用，已经安装好了3615 7.0.1-42218，
重启好几次，偶尔能进一次系统，但一直是系统可恢复的状态……

改天再试吧

exeron21, 2023-05-17:
遇到了“ 我们检测到硬盘（3，4）中有错误，因此SATA端口已被禁用”

用ARPL引导的时候是这样的：
[image]

只好把SataPortMap改成1，应该是只指定第1块硬盘，等装好了再改回来处理其他硬盘
这种办法确实能装上系统，但是现在一直是可恢复状态，断网安装也不好使

另外，主板上记录的DSM版本。。。 什么鬼，还会在主板上记录信息的吗？
[image]

wjz304, 2023-05-17:
SataPortMap 不能为1

wjz304, 2023-05-17:
3，4 是什么盘？ 他说主板 其实就是引导盘而已

exeron21, 2023-05-18:
> 3，4 是什么盘？ 他说主板 其实就是引导盘而已

主板上SATA3.0 * 2
msata * 1
minipcie * 1
esata * 1
我也不知道3，4是什么盘，也许是esata
我试过2个sata，1个msata不同的组合，一样的结果

exeron21, 2023-05-19:
> > 3，4 是什么盘？ 他说主板 其实就是引导盘而已
> 
> 主板上SATA3.0 * 2 msata * 1 minipcie * 1 esata * 1 我也不知道3，4是什么盘，也许是esata 我试过2个sata，1个msata不同的组合，一样的结果

如果3，4号盘有问题的话，能不能用SataPortMap屏蔽掉？
SataPortMap=21  DiskIdxMap=0004  应该是指只用1，2和5这三块盘吧

wjz304, 2023-05-19:
> 图片传不上来， insmod e1000e.ko insmod: can't insert 'e1000e.ko': invalid module format
> 
> find / -name e1000e.ko | xargs ls -l 发现/usr/lib/modules/6.1.26/kernel/drivers/net/ethernet/intel/e1000e/e1000e.ko 这个文件比刚才下载的要大很多 这个有473784，下载的只有：271312
> 
> 是不是引导的驱动没问题，但是安装的DSM中的驱动有问题？ 我会一点Linux和Shell，但是不懂驱动和编译，不知道该怎么手动打入驱动

你find 的是 arpl 的驱动， arpl 和 dsm 是两个系统。驱动不一样， 第十也不完全就是驱动问题，和 arpl的驱动加载机制也有关系。

wjz304, 2023-05-19:
> > > 3，4 是什么盘？ 他说主板 其实就是引导盘而已
> > 
> > 
> > 主板上SATA3.0 * 2 msata * 1 minipcie * 1 esata * 1 我也不知道3，4是什么盘，也许是esata 我试过2个sata，1个msata不同的组合，一样的结果
> 
> 如果3，4号盘有问题的话，能不能用SataPortMap屏蔽掉？ SataPortMap=21 DiskIdxMap=0004 应该是指只用1，2和5这三块盘吧

sataportmap 的每一位是一个驱动器，你就一个驱动器，不是你想拆成2个就能拆成2个的。

wjz304, 2023-06-18:
e1000e 没有问题。
https://qun.qq.com/qqweb/qunpro/share?_wv=3&_wwv=128&appChannel=share&inviteCode=20jx8dPsU2z&contentID=1m4NKs&businessType=2&from=181174&shareSource=5&biz=ka

closed 2023-06-18 by wjz304