RROrg/rr #48: 单网口 物理机装Ds920 最新版 不能识别网卡 重启后无法进入系统

BenBenPeng, 2023-05-23:
检测到 1 个网卡, 但是仅自定义了 0 个 MACs, 其他网卡将使用自身 MACs

有这么一句黄色的

我估计是 真实的Mac地址给了eth1   实际上应该是给单网卡的 eth0

我在网上发现很多和我一样的问题
N100 CPU

上一版RC1 没问题。

BenBenPeng, 2023-05-23:
检测到 1 个网卡, 但是仅自定义了 0 个 MACs, 其他网卡将使用自身 MACs

编译完以后就是这一句话 黄色的

系统都进不去 哪有别的东西可提供啊！

BenBenPeng, 2023-05-23:
[image]

重启就没有然后了

单网卡！

BenBenPeng, 2023-05-23:
其实就是只有eth0 应该吧Mac地址给 Mac0 但编译后给了Mac1 没有和eth0匹配上！

boyguyue, 2023-05-24:
我也是同样的问题，同求解决方案
[image]

wjz304, 2023-05-24:
你都说是 不能识别网卡，然而你说这么多 就是不说什么网卡？

BenBenPeng, 2023-05-25:
你看的懂么？和什么网卡有关系么？是特么给错了网卡mac地址的问题 不懂别瞎逼逼

BenBenPeng, 2023-05-25:
螃蟹1000兆，不需要额外驱动 ，之前版本用得好好的。

boyguyue, 2023-05-25:
我的是星际蜗牛B型 i211网卡

iooy, 2023-05-25:
可以尝试安装之前把BIOS里的系统使用项改为Windows8.x后再进行安装测试下，我是折腾好久都解决的。

wjz304, 2023-05-25:
> 可以尝试安装之前把BIOS里的系统使用项改为Windows8.x后再进行安装测试下，我是折腾好久都解决的。

和efi 没关系，他俩都是efi，和mac有关系，但不是 他说的问题，

Halooooooo, 2023-05-27:
..蜗牛C同样的问题

wjz304, 2023-05-27:
> ..蜗牛C同样的问题

https://github.com/wjz304/arpl-i18n/actions/runs/5097349814
 试一下

Halooooooo, 2023-05-27:
> > ..蜗牛C同样的问题
> 
> https://github.com/wjz304/arpl-i18n/actions/runs/5097349814 试一下

6啊神速 我自定义了个mac 可以没这个问题

Halooooooo, 2023-05-27:
> > ..蜗牛C同样的问题
> 
> https://github.com/wjz304/arpl-i18n/actions/runs/5097349814 试一下

老哥有没有办法开启debug模式。。我经常在这卡死。。中英文都是。。没有一个能进下一步的
。。。
[image]

wjz304, 2023-05-27:
> > > ..蜗牛C同样的问题
> > 
> > 
> > https://github.com/wjz304/arpl-i18n/actions/runs/5097349814 试一下
> 
> 老哥有没有办法开启调试模式。。我经常在这卡死。。中英文都是。。没有一个能进下一步的。。主要是不知道啥原因。。机器就僵死了 。。。 [image]

读一下黄色的字

Halooooooo, 2023-05-27:
> > > > ..蜗牛C同样的问题
> > > 
> > > 
> > > https://github.com/wjz304/arpl-i18n/actions/runs/5097349814 试一下
> > 
> > 
> > 老哥有没有办法开启调试模式。。我经常在这卡死。。中英文都是。。没有一个能进下一步的。。主要是不知道啥原因。。机器就僵死了 。。。 [image]
> 
> 读一下黄色的字

。。不是find的问题。。是僵死了。。找不到机器。。ip也ping不通。。

BenBenPeng, 2023-05-29:
更新后已解决！

78182428, 2023-06-02:
我也遇到了这个问题 ，我n5105 小主机 esxi8.0  i226网卡 ，无法打开安装界面

wjz304, 2023-06-02:
esxi 加 e1000/vmxnet3 先进去看看

> 我也遇到了这个问题 ，我n5105 小主机 esxi8.0 i226网卡 ，无法打开安装界面

ostarcn, 2023-06-03:
星际蜗牛C型 i211单网卡，安装DS920遇到了同样的问题，也是一串黄字“检测到 1 个网卡, 但是仅自定义了 0 个 MACs, 其他网卡将使用自身 MACs”，哪位大神解决了此问题

BenBenPeng, 2023-06-03:
我换换了新版以后就好了

272289834, 2023-06-07:
N100的主板是无法关闭usb关机断电的，关闭系统要断开电源，不然开机路由器就是找不到IP地址。
有大佬能在关机时关闭主板USB供电的告知一下。

wjz304, 2023-06-09:
> N100的主板是无法关闭usb关机断电的，关闭系统要断开电源，不然开机路由器就是找不到IP地址。 有大佬能在关机时关闭主板USB供电的告知一下。

bios里呢？ 啥品牌的主板？

BenBenPeng, 2023-06-09:
> N100的主板是无法关闭usb关机断电的，关闭系统要断开电源，不然开机路由器就是找不到IP地址。 有大佬能在关机时关闭主板USB供电的告知一下。

我用的是零克的小主机，没这毛病，关机就关机了，usb关机就断电。在win下 Synology下都是正常的。

closed 2023-06-22 by wjz304

wjz304, 2023-09-16:
https://github.com/wjz304/arpl-i18n/issues/219

GuoliangCui, 2023-09-20:
> > > ..蜗牛C同样的问题
> > 
> > 
> > https://github.com/wjz304/arpl-i18n/actions/runs/5097349814 试一下
> 
> 6啊神速 我自定义了个mac 可以没这个问题

怎么自定义mac ？编译之前吗？我也蜗牛C

GuoliangCui, 2023-09-20:
> 我换换了新版以后就好了

新版本具体是哪个版本？23.9.4 ？https://github.com/wjz304/arpl-i18n/releases/tag/23.9.4