RROrg/rr #50: SA6400	机型有点bug

ltdstudio, 2023-05-25:
7.2正式版 能开机进入dsm初始化，初始化重启后机器自动关机。
7.1版本   不能进入dsm web 界面。
我机型给原机型差不多 是 超微 H11SSL+7282 麻烦查看修复一下 谢谢。

ltdstudio, 2023-05-29:
非常抱歉没有叙述完整。我就最新版的7.2系统 问题 做一个描述
[image]
SA6400 64561设定完成 启动DSM安装
[image]
进入启动界面
[image]
安装完成DSM 虽然这里 Lan1 显示 error但是ping是通的
[image]
全部安装完成 重启 页面 倒数10分钟到5分钟的时候 ping通了 倒数页面开始自动刷新，然后显示 图上内容
[image]
又过了大概2分钟 左边的IPMI页面没有显示了，主机自动关机了。

wjz304, 2023-05-29:
[image]
这个时候如果能ping通，直接访问 http://<ip>:5000/
有时候会出现页面没自动跳转的情况

OrpheeGT, 2023-05-29:
Hello,

If I understood google translate, it seems I'm having the same issue on my Proxmox.

I tried to upgrade from jimmm's beta loader to yours.

After the first boot update, then the system auto shutdown.

Please find serial log window in attachment
[sa6400_shutdown.txt](https://github.com/wjz304/arpl-i18n/files/11590307/sa6400_shutdown.txt)

> # cat 103.conf 
> args: -device 'nec-usb-xhci,id=usb-bus0,multifunction=on' -drive 'file=/var/lib/vz/dump/xxxxx/arpl-i18n-sa6400.img,media=disk,format=raw,if=none,id=drive-disk-bootloader' -device 'usb-storage,bus=usb-bus0.0,port=1,drive=drive-disk-bootloader,id=usb-disk-bootloader,bootindex=999,removable=on'
> bios: ovmf
> boot:
> cores: 12
> cpu: host
> efidisk0: local-lvm:vm-103-disk-2,efitype=4m,pre-enrolled-keys=1,size=4M
> hostpci0: 0000:00:02,pcie=1
> machine: q35
> memory: 16384
> meta: creation-qemu=7.2.0,ctime=xxxxxxxxx
> name: SA6400
> net0: virtio=xx:xx:xx:xx:x:xx,bridge=vmbr0,firewall=1
> numa: 0
> ostype: l26
> sata0: local-lvm:vm-103-disk-0,size=32G
> scsihw: virtio-scsi-single
> serial0: socket
> smbios1: uuid=xxxxxxxxxxxxx
> sockets: 1
> vmgenid: xxxxxxxxxxx

OrpheeGT, 2023-05-29:
> > Hello,
> > If I understood google translate, it seems I'm having the same issue on my Proxmox.
> > I tried to upgrade from jimmm's beta loader to yours.
> > After the first boot update, then the system auto shutdown.
> > Please find serial log window in attachment [sa6400_shutdown.txt](https://github.com/wjz304/arpl-i18n/files/11590307/sa6400_shutdown.txt)
> > > # cat 103.conf
> > > args: -device 'nec-usb-xhci,id=usb-bus0,multifunction=on' -drive 'file=/var/lib/vz/dump/xxxxx/arpl-i18n-sa6400.img,media=disk,format=raw,if=none,id=drive-disk-bootloader' -device 'usb-storage,bus=usb-bus0.0,port=1,drive=drive-disk-bootloader,id=usb-disk-bootloader,bootindex=999,removable=on'
> > > bios: ovmf
> > > boot:
> > > cores: 12
> > > cpu: host
> > > efidisk0: local-lvm:vm-103-disk-2,efitype=4m,pre-enrolled-keys=1,size=4M
> > > hostpci0: 0000:00:02,pcie=1
> > > machine: q35
> > > memory: 16384
> > > meta: creation-qemu=7.2.0,ctime=xxxxxxxxx
> > > name: SA6400
> > > net0: virtio=xx:xx:xx:xx❌xx,bridge=vmbr0,firewall=1
> > > numa: 0
> > > ostype: l26
> > > sata0: local-lvm:vm-103-disk-0,size=32G
> > > scsihw: virtio-scsi-single
> > > serial0: socket
> > > smbios1: uuid=xxxxxxxxxxxxx
> > > sockets: 1
> > > vmgenid: xxxxxxxxxxx
> > 
> > 
> > Edit : shutdown seems to be triggered only as soon as I try to call web page IP:5000
> > I'am able to login from serial console/ssh prompt... navigate in folders. But as soon as I call the webpage, shutdown triggered.
> 
> ~I would like to take a look at the settings of the DSM virtual machine in your PVE~ Oh, I saw it, Because I also need to translate, I missed the picture,
> 
> Attempt to switch lkm

can we discuss privatly ?

wjz304, 2023-05-29:
I tried it and got the same result as you,
But the recent modifications are not related to it, I need to study it.

ltdstudio, 2023-05-29:
> 物理机？

是的 是 物理机。

wjz304, 2023-05-29:
> > 物理机？
> 
> 是的 是 物理机。

你也切换lkm试试吧， 我要再测试一下，

wjz304, 2023-05-29:
I have checked the recent modifications and found the reason,
Will be fixed in the next version. I'm sorry for the trouble I've caused you.
---
我检查了最近的修改，并找到了原因，
将在下一个版本修复。 给你们带来的麻烦我很抱歉。

wjz304, 2023-05-29:
update 23.5.10 and update addons.