RROrg/rr #54: arpl-i18n 23.5.9 error

azchoi, 2023-05-29:
I've updated load 23.5.8 to 23.5.9 via update menu, I met some error message like belows. But booting fuctions is OK.
Please let me know what is this error message.
DSM version is latest 7.2 @ 3622

err 1 : can't open 'proc/meminfo' : No such file or directory
err 2 :  line 209 error of init.sh

[image]

wjz304, 2023-05-29:
Incomplete loading of the USB drive system, possibly causing damage to the ARPL system during the upgrade process

Reboot and take a look, or force upgrade again from the upgrade menu.

azchoi, 2023-05-29:
> Incomplete loading of the USB drive system, possibly causing damage to the ARPL system during the upgrade process
> 
> Reboot and take a look, or force upgrade again from the upgrade menu.

Thank you. After force usb reinstall, the error message has been gone.

closed 2023-05-29 by wjz304