RROrg/rr #58: Stick not booting

mandgeo, 2023-05-31:
Until recently I was recently using TCRP on my HP Microserver gen8. I've migrated to ARPL by fbelavenuto just to notice that I can't update to DSM 7.2.

Now I want to install the latest version of ARPL-i18n, but it's not booting. Didn't changed anything in bios. "non-system disk or disk error". What am I doing wrong? Tried to write image with balenaEtcher, Rufus, event version 23.5.9, same result.

wjz304, 2023-05-31:
I have modified grub due to some machines reporting grub errors during non EFI,

This may have caused some other incompatibilities, 
If possible, I suggest using EFI mode,

wjz304, 2023-05-31:
https://github.com/wjz304/arpl-i18n/releases/download/23.5.10/arpl-i18n-23.5.10.img-gen8-test.zip

I made a small modification, but I'm not sure if it works,
If it's convenient for you, please test it

mandgeo, 2023-05-31:
It works with modified version. 
Future ARPL updates will result in boot problems?

Thank you!

wjz304, 2023-06-01:
从逻辑上不会影响更新，因为这个修改是修改了分区属性，并没实际的文件修改，arpl的 "update menu" 目前只能做到文件更细，并不包含磁盘属性的更新。

Logically, it will not affect the update, as this modification modifies partition properties without actual file modifications. Currently, ARPL's "update menu" can only achieve finer file details and does not include updates to disk properties.

However, in order to help more people solve this problem, I will release 23.6.1 later and suggest that you use it.

wjz304, 2023-06-01:
https://github.com/wjz304/arpl-i18n/releases/tag/23.6.0

closed 2023-06-02 by wjz304