RROrg/rr #59: 23.6.0 无法登陆Synology 账号

yaocanfeng, 2023-06-01:
23.5.10-23.6.0 两个引导都测试过，在面板的 Synology 账户 点击登录，无法弹出 登录页面，退回到23.5.8  的引导就可以

wjz304, 2023-06-02:
报什么错？

yaocanfeng, 2023-06-02:
没报错，点击后看到鼠标转一下就没反应，无法弹出登录页面。

wjz304, 2023-06-02:
换个浏览器试试，我可以正常弹出

yaocanfeng, 2023-06-02:
换过，edge 谷歌 火狐，清理过浏览器缓存，也更试过用电脑，平板，去登录都不行。

wjz304, 2023-06-02:
那就先用旧版吧，我只能尝试复现一下

wjz304, 2023-06-02:
[image]

wjz304, 2023-06-02:
@yaocanfeng  自定义下mac，应该就可以了

我再研究通用的方法...

yaocanfeng, 2023-06-02:
sn 和mac 是对的。

wjz304, 2023-06-02:
设置了sn和mac么？ 那就不知道了，

刚有一个同样问题的，没有设置mac的，调试了一下，设置了mac就好了

yaocanfeng, 2023-06-02:
mac 不对只是不能授权，登录还是可以的吧。

wjz304, 2023-06-02:
23.5.9 我修改了如果没有自定义mac，cmdline 会不传mac，
23.5.8 如果没有自定义mac，会抓取网卡mac 传下去，
就这点区别

取消往下传mac 是因为 一部分情况下的一些网卡，设置了mac会dhcp出错 导致搜不到设备，即使传原始mac也会dhcp出错。

yaocanfeng, 2023-06-02:
我是虚拟机，直接在虚拟机改了网卡mac，不是从cmdline  自定义的。

wjz304, 2023-06-02:
cmdlime 传一下吧，再menu 设置自定义mac ，默认他会抓你设置的mac 直接确认就好了

yaocanfeng, 2023-06-02:
cmdlime  我没成功过，每次自定义都获取不到IP。有空我再测试下。

wzx-moe, 2023-06-04:
有同样问题，原本登录好的， 升级完 账户 qc ddns 全部不能用了

a8223492, 2023-06-05:
> 有同样问题，原本登录好的， 升级完 账户 qc ddns 全部不能用了

我也有同样问题。

yaocanfeng, 2023-06-05:
> 
所以没事别轻易升级。

gdzjy, 2023-06-06:
一开始也是不行，后来在引动编译界面 定义一下MAC地址（其实就是回车确定就可以了，因为引导自动读取了网卡MAC）就可以了。

closed 2023-06-12 by wjz304