RROrg/rr #60: r8125的网卡系统引导以后就显示是r8169，提示IP地址错误，最终就是失联了

xinjing-cloud, 2023-06-02:
r8125的网卡系统引导以后就显示是r8169，提示IP地址错误，最终就是失联了

wjz304, 2023-06-03:
dhcp 问题。

Jial08, 2023-06-04:
> dhcp 问题。

大佬，能详细点说说怎么解决吗？我也是这个问题，r8125b的网卡识别为r8169，还总是在开机的时候识别 IP error 导致开不了机，偶发性的，感谢
[image]

wjz304, 2023-06-04:
> > dhcp 问题。
> 
> 大佬，能详细点说说怎么解决吗？我也是这个问题，r8125b的网卡识别为r8169，还总是在开机的时候识别 IP error 导致开不了机，偶发性的，感谢 [image]

在模块中取消 r8169，然后重新编译，(不用管 arpl的显示，只是提示性信息)，

Jial08, 2023-06-04:
好的，感谢，我试试

naoki66, 2023-06-06:
谢谢 我也是b360+8600t 网卡8125b  取消了r8169选r8125 重新编译后 还是失联

CarlosLucky, 2023-06-06:
> 

你就一块网卡还是多个网卡啊

naoki66, 2023-06-06:
> 谢谢 我也是B360-M-AORUS-PRO+8600t 网卡8125b(4口）+1个板载intel 219 取消了r8169选r8125 重新编译后 还是失联，全选models也不行。只能在arpl界面获取网卡IP，显示启动到dsm后IP就ping不通（但是网卡灯亮），也找不到其他IP。

使用基于arpl-i18n 23.5.8旧版本的gxnas引导盘成功

limitget, 2023-06-09:
同样遇到

wjz304, 2023-06-18:
r8125 有问题请重新跟反馈 issue https://github.com/wjz304/arpl-i18n/issues/84

closed 2023-06-18 by wjz304