RROrg/rr #61: Can you solve the Unknown symbol issue of alx.ko (atheros)?

PeterSuh-Q3, 2023-06-03:
SynologyNAS> dmesg | grep alx
[   35.023215] alx: Unknown symbol mdio_mii_ioctl (err 0)

As a result of processing with modprobe or insmod in the DDSML method of TCRP, an Unknown symbol message came out.
As a result of using ARPL's EUDEV method, an Unknown symbol message appeared in dmesg as shown below.

Because of this, modules for DSM 7.2 will not work.
On my 2nd XPE the alx nic became unavailable.
Any solution?
Does the source of the alx directory in the src path need to be modified compared to the Linux source?

wjz304, 2023-06-03:
哪个型号？

wjz304, 2023-06-03:
你测试的型号是这4个中的吗？
[image]

PeterSuh-Q3, 2023-06-03:
thank you First tried with r1000 (DS923+).
I'll fix it as you suggested and recompile it.

PeterSuh-Q3, 2023-06-03:
By the way, my source was already pointing to the CONFIG_MDIO=n value.
There is nothing to fix.

[image]

PeterSuh-Q3, 2023-06-03:
r1000 is not available because CONFIG_MDIO=n
If CONFIG_MDIO=m of geminilake, is NIC operation possible?

wjz304, 2023-06-03:
I don't have this network card, so I'm not sure if it's possible. You can try the model of geminilake.
For what reason fbelavenuto closed r1000/v1000/.., it is currently unknown, but we can open it and compile it

PeterSuh-Q3, 2023-06-03:
thank you I tried the DS920+ first and
DS923+, as you said, open r1000 with m value, recompile and try again.

PeterSuh-Q3, 2023-06-03:
This is the test result of alx.ko.

As a result of using the original version of your geminilake alx.ko, it worked normally.
Apollolake doesn't seem to have a problem either.

As instructed, I activated mdio on 5 platforms and compiled the entire module.
The following changes were confirmed
r1000, denverton obviously have mdio enabled, but they are not included in the compilation target.
It's strange that all geminilake was recompiled unnecessarily.

[image]

Tried DS1621+ for v1000 compiled after enabling mdio, it doesn't work with unknown symbol message.

I haven't tried all the other platforms, but they seem to be the same.

wjz304, 2023-06-03:
I also compiled it through Action, and you can try to extract the corresponding platform's mdio.ko, `insmod mdio.ko`
https://github.com/wjz304/arpl-modules/actions/runs/5163988066

PeterSuh-Q3, 2023-06-04:
I imported and used mdio.ko for DSM 7.2 from pocopico.
His mdio.ko was already responding to all platforms.

As a result of testing v1000 as a sample, all of them were automatically activated in eudev / ddsml just by including mdio.ko in the integrated module pack tgz.

https://github.com/PeterSuh-Q3/arpl-modules/releases/tag/v1.33

I've also included mdio.ko on all platforms and now expect all unknown symbol issues to be gone.

https://github.com/PeterSuh-Q3/arpl-modules/tree/main/vanilla

thank you

closed 2023-06-04 by PeterSuh-Q3

PeterSuh-Q3, 2023-06-04:
The modules that depend on mdio.ko have been identified as the 4 nic modules below.

filename: /root/arpl-modules/v1000-4.4.302/alx.ko Qualcomm Atheros(R) AR816x/AR817x PCI-E Ethernet
filename: /root/arpl-modules/v1000-4.4.302/cxgb.ko Chelsio 10Gb Ethernet
filename: /root/arpl-modules/v1000-4.4.302/cxgb3.ko Chelsio T3 Network
filename: /root/arpl-modules/v1000-4.4.302/sfc.ko Solarflare SFC4000/SFC9000/SFC9100-family 10/40-gigabit Ethernet