RROrg/rr #62: sa6400识别不到硬盘

sjkhsl, 2023-06-04:
3622、4021、918都可以装，就sa6400死活识别不了硬盘！

xintonglaile, 2023-06-04:
借楼问一下后面会支持sata硬盘做引导吗？现在sa6400只支持u盘引导哎。

wjz304, 2023-06-05:
sata 也能装，但是流程目前不友好

sjkhsl, 2023-06-05:
> sata 也能装，但是流程目前不友好

坐等大佬更新～

sjkhsl, 2023-06-05:
> sata 也能装，但是流程目前不友好

大佬，23.6.1引导sa6400还是找不到硬盘！pve下。

wjz304, 2023-06-05:
> > sata 也能装，但是流程目前不友好
> 
> 大佬，23.6.1引导sa6400还是找不到硬盘！pve下。

6.1 我也没提和 sa6400 有关的修改啊...

别人都找的到硬盘，你找不到硬盘，啥也没有，我把手指头掐破了也不知道你啥环境啊。

sjkhsl, 2023-06-05:
> > > sata 也能装，但是流程目前不友好
> > 
> > 
> > 大佬，23.6.1引导sa6400还是找不到硬盘！pve下。
> 
> 6.1 我也没提和 sa6400 有关的修改啊...
> 
> 别人都找的到硬盘，你找不到硬盘，啥也没有，我把手指头掐破了也不知道你啥环境啊。

pve7.4，给群晖直通了5块盘，一块ssd，4块机械，也尝试pve分配一块200G硬盘，提示找不到硬盘。

wjz304, 2023-06-18:
环境不明确，没有日志，无法继续分析

ltdstudio, 2023-06-23:
> 借楼问一下后面会支持sata硬盘做引导吗？现在sa6400只支持u盘引导哎。

同求

wjz304, 2023-06-23:
jim大已经做完了，看他博客吧

minglihui998, 2023-08-02:
ARPL-i18n 23.7.6      SA6400  7.2 无法识别Lsi2308 直通卡下面的硬盘， 能识别到我主板Sata口的硬盘， CPU E31240L V5，主板话说P10s_E/4L(C236芯片组），内存 三星64G（16GX4）纯ECC 2133，LSI2308阵列卡刷直通，接4块Sas4T +4块Sata4T ,主板sata 接intel 800G 固态（S3520）,同配置安装3622XS+没有问题，安装918+也没有问题。

wjz304, 2023-08-10:
DT 型号目前不支持sas扩展

closed 2023-08-10 by wjz304