RROrg/rr #63: Guide to essential vanilla modules for console addon to work in DSM 7.2

PeterSuh-Q3, 2023-06-04:
Starting with mdio.ko you pointed out, I checked more modules for missing dependencies.
Most of them were modules that had to be imported from vanilla modules.
mdio had 4 nics and dependencies
Six vanilla modules for fabio's console addon to work were missing on the two platforms below.

cfbcopyarea.ko
cfbfillrect.ko
cfbimgblt.ko
fb.ko
fb_sys_fops.ko
fbdev.ko

After applying this module, the console (cgetty of M Shell) works normally.

https://github.com/PeterSuh-Q3/arpl-modules/tree/main/vanilla/apollolake-4.4.302

https://github.com/PeterSuh-Q3/arpl-modules/tree/main/vanilla/geminilake-4.4.302

wjz304, 2023-06-04:
Thank you for sharing
arpl's modules. tgz is directly decompressed to /usr/lib/modules/ of rd.gz, so what already exists in rd.gz does not actually need to be included in modules. tgz

wjz304, 2023-06-04:
Oh, I seem to have misread it. them're not in rd.gz
[image]

PeterSuh-Q3, 2023-06-05:
Could you please share with me the command to check dependencies like above?

wjz304, 2023-06-05:
```
workspace=$PWD
for M in `ls ${workspace}`; do
  depends=(`modinfo "${workspace}/${M}" | grep depends: | awk -F: '{print $2}' | awk '$1=$1' | sed 's/,/ /g'`)
  if [ ${#depends[*]} -gt 0 ]; then
    for k in ${depends[@]}; do
      [ ! -f "${workspace}/${k}" ] && echo "[W] '${M}' depends '${k}' does not exist" 
    done
  fi
done
```

wjz304, 2023-06-05:
https://github.com/wjz304/arpl-modules/blob/main/.github/workflows/main.yml
line: 278

PeterSuh-Q3, 2023-06-05:
thank you I made some adjustments to make it usable in the Github Codespaces Ubuntu terminal environment.
It works fine.

@PeterSuh-Q3 ➜ /workspaces/arpl-modules/apollolake-4.4.302 (main) $ cat *.sh
#!/usr/bin/env bash

for M in `ls *.ko`; do
   depends=`modinfo "${M}" | grep depends: | awk -F: '{print $2}' | awk '$1=$1' | sed 's/,/ /g'`
   if [ ${#depends[*]} -gt 0 ]; then
     for k in `echo "$depends"`; do
       [ ! -f "${k}.ko" ] && echo "[W] ${M} depends ${k} does not exist"
     done
   fi
done

@PeterSuh-Q3 ➜ /workspaces/arpl-modules/apollolake-4.4.302 (main)$sh chk.sh
[W] acpi-cpufreq.ko depends processor does not exist
[W] atlantic.ko depends crc-itu-t does not exist
[W] b44.ko depends ssb does not exist
[W] dvb-usb-dib0700.ko depends dib7000m does not exist
[W] dvb-usb-dib0700.ko depends dib9000 does not exist
[W] dvb-usb-dib0700.ko depends dibx000_common does not exist
[W] dvb-usb-dib0700.ko depends dib0090 does not exist
[W] dvb-usb-dib0700.ko depends dib0070 does not exist
[W] dvb-usb-dib0700.ko depends dib3000mc does not exist
[W] m88ds3103.ko depends regmap-i2c does not exist
[W] r8153_ecm.ko depends cdc_ether does not exist
[W] ts2020.ko depends regmap-i2c does not exist
[W] usbcore.ko depends usb-common does not exist
[W] vxlan.ko depends udp_tunnel does not exist
[W] vxlan.ko depends ip6_udp_tunnel does not exist
@PeterSuh-Q3 ➜ /workspaces/arpl-modules/apollolake-4.4.302 (main)$

wjz304, 2023-06-05:
yes, there are some differences between different platforms, and some dependencies are not easy to solve

closed 2023-06-06 by PeterSuh-Q3