RROrg/rr #67: 求助：无法定时开机

zhuzizhao1, 2023-06-07:
你好，大佬，非常感谢你制作出如此方便的汉化编译工具，在使用过程中我遇到个问题，还请您方便的时候予以帮助。问题是我使用的最新版本的引导，黑裙机器硬件是g4400t处理器搭配微星h110m pro d主板，在使用dsm7.2引导后无法定时开机，而在这之前我使用的是jun的6.2.3的引导，这个6.2.3系统可以实现定时开机，个人习惯了定时开机、定时关机，请大神帮我分析下，有什么办法既能体验新系统还能使用dsm内置的定时开机功能呢。

zhuzizhao1, 2023-06-07:
[image]

wjz304, 2023-06-10:
看一下bios时间

zhuzizhao1, 2023-06-11:
我有一张这两天主板bios的截图，截图显示，bios时间比实际时间早了8个小时，我记得dsm7.0刚出来的时候我图新鲜刷过dsm7.0也存在这个问题，当时尝试过修改主板时间但是没有用，每次只要黑裙一开机，群晖系统时间都是正常的，bios都会快8小时，当时以为是引导的问题，就一直等到现在的引导，但是还是存在这个问题。

loveu5522, 2023-06-14:
我也是同样问题不知道怎么回事

gbm777, 2023-07-22:
是不是群晖的时区不对

hudzy, 2023-09-21:
Same here, J4125 + DS920+ + DSM 7.2, ARPL-i18N version [23.9.0](https://github.com/wjz304/arpl-i18n/releases/tag/23.9.0)

BIOS timezone is UTC and DMS TZ is CST

LD166, 2023-11-24:
N100 CPU 安装SA3622系统 可以在系统里正常设置开关机功能，但是换SA6400 就不行了，开启SA6400 的 SSH 使用 dmesg | grep -i rtc 查看rtc 信息，提示没有pnp设备,如下：

wjz304, 2024-03-30:
powersched addon

closed 2024-04-19 by wjz304

jiangyh91, 2024-07-17:
相同问题，J5005 & DS918+ & DSM 7.2 & rr version 24.6.9
BIOS显示UTC时区，DMS系统显示使用用户正确时区

**安装powersched插件后，完美解决**