RROrg/rr #69: SA6400 AME with latest 7.2 64570

OrpheeGT, 2023-06-09:
Hello,

Latest DSM update fails to launch Advanced Media Extension.

I have an error at application launch.

And in synoscgi.log I have : 

```
synoscgi_SYNO.AME.License_1_check[00000]: boot/boot_lock.c(228): failed to mount boot device /dev/synoboot2 /tmp/bootmnt (errno:13)
synoscgi_SYNO.AME.License_1_check[00000]: utils.cpp:18 init fail
synoscgi_SYNO.AME.License_1_check[00000]: boot/boot_lock.c(228): failed to mount boot device /dev/synoboot2 /tmp/bootmnt (errno:13)
synoscgi_SYNO.AME.License_1_check[00000]: license_manager.cpp:34 Failed to get "uuid" from "SLGetUUID()"
synoscgi_SYNO.AME.License_1_check[00000]: license.cpp:39 Failed to create LicenseManger
synoscgi_SYNO.Core.MyDSCenter.Login_1_pkce_config_get[00000]: dsinfo_cloudservice_mac_get.c:80 cannot get the mac address from vender
synoscgi_SYNO.Core.MyDSCenter.Login_1_pkce_config_get[00000]: dsm_service.cpp:35 Failed to get mac address
synoscgi_SYNO.Entry.Request_1_request[00000]: APIUtilsPrivate.cpp:83 protocol failed
synoscgi_SYNO.Entry.Request_1_request[00000]: CompoundHandler.cpp:342 read compound json failed
```

It seems I don't have this issue on DVA3221 loader.

OrpheeGT, 2023-06-09:
So actually, I had to force define a "custom MAC" in the loader, even if the MAC is same as original host.
Probably due to the recent change.

closed 2023-06-09 by OrpheeGT

wjz304, 2023-06-10:
Yes, in some cases, effective IP can only be obtained without passing in the MAC. 
Currently, I do not have a good solution

OrpheeGT, 2023-06-10:
> Yes, in some cases, effective IP can only be obtained without passing in the MAC. Currently, I do not have a good solution

if there are only some modules/ext (as vmxnet3, r8125) would it be possible to add a condition to disable custom MAC only when these kind of modules are loaded ?

wjz304, 2023-06-10:
> > Yes, in some cases, effective IP can only be obtained without passing in the MAC. Currently, I do not have a good solution
> 
> if there are only some modules/ext (as vmxnet3, r8125) would it be possible to add a condition to disable custom MAC only when these kind of modules are loaded ?

There is this consideration, but the specific criteria for judgment are currently not very clear

OrpheeGT, 2023-06-10:
Why would someone want to use vmxnet3 on PVE ?! :D LOL virtio or nothing :p

wjz304, 2023-06-10:
> Why would someone want to use vmxnet3 on PVE ?! :D LOL virtio or nothing :p

Because virtio may not work in PVE, such as me,
[image]

wjz304, 2023-06-10:
What version is your PVE? Is it 7.4-3?

OrpheeGT, 2023-06-10:
[image]

I always keep it up to date.

wjz304, 2023-06-10:
My impression is that virtio cannot be work is from the version of 7.3, and I haven't had time to delve deeper
My current version is 7.4-3, I'll upgrade it tonight and take a look.
[image]

OrpheeGT, 2023-06-10:
I never had any issues with virtio ...  running since 7.1 keeping up to date each time a new release available.

OrpheeGT, 2023-06-10:
Host details :

```
00:00.0 Host bridge [0600]: Intel Corporation 8th/9th Gen Core 8-core Desktop Processor Host Bridge/DRAM Registers [Coffee Lake S] [8086:3e30] (rev 0a)
        Subsystem: Super Micro Computer Inc 8th/9th Gen Core 8-core Desktop Processor Host Bridge/DRAM Registers [Coffee Lake S] [15d9:1a1d]
        Flags: bus master, fast devsel, latency 0, IOMMU group 0
        Capabilities: [e0] Vendor Specific Information: Len=10 <?>
        Kernel driver in use: skl_uncore
        Kernel modules: ie31200_edac

00:01.0 PCI bridge [0604]: Intel Corporation 6th-10th Gen Core Processor PCIe Controller (x16) [8086:1901] (rev 0a) (prog-if 00 [Normal decode])
        Flags: bus master, fast devsel, latency 0, IRQ 122, IOMMU group 1
        Bus: primary=00, secondary=01, subordinate=01, sec-latency=0
        I/O behind bridge: 00006000-00006fff [size=4K]
        Memory behind bridge: 70000000-830fffff [size=305M]
        Prefetchable memory behind bridge: [disabled]
        Capabilities: [88] Subsystem: Super Micro Computer Inc 6th-9th Gen Core Processor PCIe Controller (x16) [15d9:1a1d]
        Capabilities: [80] Power Management version 3
        Capabilities: [90] MSI: Enable+ Count=1/1 Maskable- 64bit-
        Capabilities: [a0] Express Root Port (Slot+), MSI 00
        Capabilities: [100] Virtual Channel
        Capabilities: [140] Root Complex Link
        Capabilities: [d94] Secondary PCI Express
        Kernel driver in use: pcieport

00:01.1 PCI bridge [0604]: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor PCIe Controller (x8) [8086:1905] (rev 0a) (prog-if 00 [Normal decode])
        Flags: bus master, fast devsel, latency 0, IRQ 123, IOMMU group 2
        Bus: primary=00, secondary=02, subordinate=02, sec-latency=0
        I/O behind bridge: 00005000-00005fff [size=4K]
        Memory behind bridge: 50000000-630fffff [size=305M]
        Prefetchable memory behind bridge: [disabled]
        Capabilities: [88] Subsystem: Super Micro Computer Inc Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor PCIe Controller (x8) [15d9:1a1d]
        Capabilities: [80] Power Management version 3
        Capabilities: [90] MSI: Enable+ Count=1/1 Maskable- 64bit-
        Capabilities: [a0] Express Root Port (Slot+), MSI 00
        Capabilities: [100] Virtual Channel
        Capabilities: [140] Root Complex Link
        Capabilities: [d94] Secondary PCI Express
        Kernel driver in use: pcieport

00:02.0 Display controller [0380]: Intel Corporation CoffeeLake-S GT2 [UHD Graphics 630] [8086:3e98]
        Subsystem: Super Micro Computer Inc UHD Graphics 630 (Desktop 9 Series) [15d9:1a1d]
        Flags: fast devsel, IRQ 11, IOMMU group 3
        Memory at 84000000 (64-bit, non-prefetchable) [disabled] [size=16M]
        Memory at 40000000 (64-bit, prefetchable) [disabled] [size=256M]
        I/O ports at 7000 [disabled] [size=64]
        Capabilities: [40] Vendor Specific Information: Len=0c <?>
        Capabilities: [70] Express Root Complex Integrated Endpoint, MSI 00
        Capabilities: [ac] MSI: Enable- Count=1/1 Maskable- 64bit-
        Capabilities: [d0] Power Management version 2
        Capabilities: [100] Process Address Space ID (PASID)
        Capabilities: [200] Address Translation Service (ATS)
        Capabilities: [300] Page Request Interface (PRI)
        Kernel driver in use: vfio-pci
        Kernel modules: i915

00:08.0 System peripheral [0880]: Intel Corporation Xeon E3-1200 v5/v6 / E3-1500 v5 / 6th/7th/8th Gen Core Processor Gaussian Mixture Model [8086:1911]
        Subsystem: Super Micro Computer Inc Xeon E3-1200 v5/v6 / E3-1500 v5 / 6th/7th/8th Gen Core Processor Gaussian Mixture Model [15d9:1a1d]
        Flags: fast devsel, IRQ 11, IOMMU group 4
        Memory at 400011d000 (64-bit, non-prefetchable) [disabled] [size=4K]
        Capabilities: [90] MSI: Enable- Count=1/1 Maskable- 64bit-
        Capabilities: [dc] Power Management version 2
        Capabilities: [f0] PCI Advanced Features

00:12.0 Signal processing controller [1180]: Intel Corporation Cannon Lake PCH Thermal Controller [8086:a379] (rev 10)
        Subsystem: Super Micro Computer Inc Cannon Lake PCH Thermal Controller [15d9:1a1d]
        Flags: fast devsel, IRQ 16, IOMMU group 5
        Memory at 400011c000 (64-bit, non-prefetchable) [size=4K]
        Capabilities: [50] Power Management version 3
        Capabilities: [80] MSI: Enable- Count=1/1 Maskable- 64bit-
[... 396 lines cut ...]
-rw-r--r-- 1 root root 137 Feb 26 23:46 blacklist.conf
-rw-r--r-- 1 root root 154 Mar  2  2022 intel-microcode-blacklist.conf
-rw-r--r-- 1 root root  48 Apr  5 16:26 kvm.conf
-rw-r--r-- 1 root root 172 May 28 19:54 pve-blacklist.conf
-rw-r--r-- 1 root root  45 Feb 26 23:55 vfio.conf
root:/etc/modprobe.d# cat *
blacklist amdgpu
blacklist radeon
blacklist nouveau
blacklist nvidia
blacklist snd_hda_intel
blacklist snd_hda_codec_hdmi
blacklist i915
# The microcode module attempts to apply a microcode update when
# it autoloads.  This is not always safe, so we block it by default.
blacklist microcode
options kvm ignore_msrs=1 report_ignored_msrs=0
# This file contains a list of modules which are not supported by Proxmox VE

# nvidiafb see bugreport https://bugzilla.proxmox.com/show_bug.cgi?id=701
blacklist nvidiafb
options vfio-pci ids=8086:3e98 disable_vga=1
```

This is my personnal settings to be able to passthrough Nvidia GTX1650 and also iGPU intel HD630.

OrpheeGT, 2023-06-10:
and my grub settings too :

```
root@:/etc/default# cat grub
GRUB_DEFAULT=0
GRUB_TIMEOUT=5
GRUB_DISTRIBUTOR=`lsb_release -i -s 2> /dev/null || echo Debian`
GRUB_CMDLINE_LINUX_DEFAULT="quiet"
GRUB_CMDLINE_LINUX="intel_iommu=on iommu=pt ipv6.disable=1"
```

wjz304, 2023-06-10:
Yes, I suspect it is related to my physical network card, so I haven't delved deeper,
My physical network card is r8125, and I have also recompiled the driver in PVE,

OrpheeGT, 2023-06-10:
```
root:/etc/network# ethtool -i eno2
driver: igb
version: 5.15.107-2-pve
firmware-version: 3.25, 0x800005cc
expansion-rom-version:
bus-info: 0000:07:00.0
supports-statistics: yes
supports-test: yes
supports-eeprom-access: yes
supports-register-dump: yes
supports-priv-flags: yes

root:/etc/network# ethtool eno2
Settings for eno2:
        Supported ports: [ TP ]
        Supported link modes:   10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
        Supported pause frame use: Symmetric
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
        Advertised pause frame use: Symmetric
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Speed: 1000Mb/s
        Duplex: Full
        Auto-negotiation: on
        Port: Twisted Pair
        PHYAD: 1
        Transceiver: internal
        MDI-X: off (auto)
        Supports Wake-on: pumbg
        Wake-on: g
        Current message level: 0x00000007 (7)
                               drv probe link
        Link detected: yes
```

```
root:/etc/network# cat interfaces

auto lo
iface lo inet loopback

iface eno2 inet manual
        ethernet-wol g

iface eno1 inet manual
        ethernet-wol g

auto vmbr0
iface vmbr0 inet static
        address 192.168.x.x/24
        gateway 192.168.x.x
        bridge-ports eno2
        bridge-stp off
        bridge-fd 0
```

wjz304, 2023-06-10:
Upgrading PVE to 7.4-13, A magical scene，virtio working
[image]

wjz304, 2023-06-12:
dditional explanation,
It should not be directly related to the PVE version.
Because I am r8125 and have compiled the driver myself,
Therefore, the driver should be recompiled after each PVE upgrade