RROrg/rr #70: 配置之后DSM启动找不到网卡

cuitzeno, 2023-06-09:
刷了最新的arpl-i18n-23.6.2.img.zip，插上U盘配置好，DSM启动之后无法找到IP。。。不知道是不是我的配置有问题。请帮忙查看一下，谢谢。
[image]

cuitzeno, 2023-06-09:
主板是畅网妖板N5105，4个2.5G网口。之前运行7.1正常，今天想升级，结果出问题了。。。

limitget, 2023-06-09:
我升级也是遇到这个情况

limitget, 2023-06-09:
我现在是7.2的系统，用别人的引导安装上的，用I18n就没办法启动，显示网卡错误

cuitzeno, 2023-06-09:
谁的系统？我也去试试呢。。。

wjz304, 2023-06-09:
正好我手里有这个板子，刚从sa6400 7.1.1 迁移到 920 7.2 没出现你说的情况，
就是等的时间相对久一点.

关于ip显示错误的问题，不想再说了，介意就用其他引导吧

cuitzeno, 2023-06-10:
后面重新安装了923，把之前的盘迁移过去了。但是虚拟机不知道怎么回事无法启动创建的虚拟机

wjz304, 2023-06-10:
> 后面重新安装了923，把之前的盘迁移过去了。但是虚拟机不知道怎么回事无法启动创建的虚拟机

923 是amd的u

cuitzeno, 2023-06-10:
那我要用NVMe当存储该选什么型号？谢谢。

cuitzeno, 2023-06-10:
之前装920升级7.2，NVME的存储说不受系统支持

wjz304, 2023-06-10:
> 之前装920升级7.2，NVME的存储说不受系统支持

有补丁，

wjz304, 2023-06-10:
https://github.com/007revad/Synology_enable_M2_volume

cuitzeno, 2023-06-15:
问题全部解决了，换回了920+，谢谢大佬。。。