RROrg/rr #79: i5-10500 核显直通不成功

z886, 2023-06-11:
首先感谢大神提供的引导，但在ESXI7.0中，直通核显不成功，麻烦大神指导下，谢谢。
宿主机：华硕主板B150M-PLUS
虚拟机版本：ESXI7.0
引导版本：arpl-i18n-23.6.2.vmdk-flat
DSM版本：DS918+ 7.1.1
直通了网卡和核显，也在编译引导时，添加了外部插件 i915-10th.tar.gz ，安装完成后，进入DSM，用root登录SSH下敲命令 ls /dev/dri ，显示为空，核显没有直通成功；之前用的传统引导，DS918+ 7.0.1 核显是正常的。

myworksfun, 2023-06-12:
10代的U好像就不支持核显直通，esxi和PVE均不行。。

naoki66, 2023-06-17:
> 10代的U好像就不支持核显直通，esxi和PVE均不行。。

PVE下我测试SeaBIOS，rombar=0，DS918正常

z886, 2023-06-22:
> > 10代的U好像就不支持核心显示直通，esxi和PVE均不行。。
> 
> PVE下我测试SeaBIOS，rombar=0，DS918正常 [image]

cat /sys/kernel/debug/dri/0/i915_frequency_info

> > 10代的U好像就不支持核心显示直通，esxi和PVE均不行。。
> 
> PVE下我测试SeaBIOS，rombar=0，DS918正常 [image]
ESXI7.0，DSM 7.0.1-42218 Update 3 ，用的传统引导，可以直通的。
[image]

z886, 2023-06-22:
> > 10代的U好像就不支持核心显示直通，esxi和PVE均不行。。
> 
> PVE下我测试SeaBIOS，rombar=0，DS918正常 [image]

ESXI7.0，DSM 7.0.1-42218 Update 3 ，用的传统引导，可以直通的。
[image]

wjz304, 2023-06-28:
试最新的把，

closed 2023-06-28 by wjz304

tanst, 2023-10-08:
> > 10代的U好像就不支持核显直通，esxi和PVE均不行。。
> 
> PVE下我测试SeaBIOS，rombar=0，DS918正常 [image]

我用 q35 能直通核显，但是在群晖里面查看 `lspci -n` 核显的pci编号就变成 0000:00:01.0 0300: 8086:3e91
这样 01 了，后果就是无法使用硬解，但是能看到核显
cat /sys/kernel/debug/dri/0/i915_frequency_info
也能正常显示支持硬解