RROrg/rr #80: aprl-i18n 23.6.3 找不到IP

cliveming, 2023-06-11:
I am using Baremetal with NIC Realtek 8169,
There was no issue with aprl-i18n 23.5.8, with DSM 7.2 64561.

Trying to update loader version to 23.6.3, and NIC unable to get IP.

Manually downgrade to arpl-i18n 23.6.2, with DSM7.2 64570, and working good.

wjz304, 2023-06-11:
Yes, this is a very headache issue. 
The driver of 8169 is compatible with 8125, which has led to too many people coming to me to say that r8125 is recognized as 8169. I have explained it too many times without affecting it, but still many people talk about this issue every day. So I blocked the driver of 8169,

You can use 23.6.2 first, I'll think of other ways.

wjz304, 2023-06-11:
Or for now, you can, in the shell of arpl, delete the shielding file of r8169
```
RDXZ_PATH=/tmp/rdxz_tmp
mkdir -p "${RDXZ_PATH}"
(cd "${RDXZ_PATH}"; xz -dc < "/mnt/p3/initrd-arpl" | cpio -idm) >/dev/null 2>&1 || true
rm -rf "${RDXZ_PATH}/etc/modprobe.d/blacklist-r8169.conf"
(cd "${RDXZ_PATH}"; find . 2>/dev/null | cpio -o -H newc -R root:root | xz --check=crc32 > "/mnt/p3/initrd-arpl") || true
rm -rf "${RDXZ_PATH}"
reboot
```

wjz304, 2023-06-11:
If convenient, plz take a screenshot of the ID of the network card
```
root@arpl:/opt/arpl# lspci -nn
06:10.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller [10ec:8125] (rev 05)
```

limitget, 2023-06-11:
[image]
我的是这个版本，

wjz304, 2023-06-11:
https://github.com/wjz304/arpl-i18n/releases/tag/23.6.4

closed 2023-06-11 by wjz304

PeterSuh-Q3, 2023-06-11:
I referred to the source of the realtek nic module pocopico.
The directories for each module are all divided,
I haven't checked whether r8125 and r8169 can not affect each other.
In my test environment, these two chipsets cannot be tested together on one PC.
If the environment is the same, we will share the test results.

https://github.com/PeterSuh-Q3/arpl-modules/tree/main/src/4.x/drivers/net/ethernet/realtek