RROrg/rr #81: 为什么升级到23.6.3 编译的主面板都进不去了

limitget, 2023-06-11:
[image]
[image]
[image]

PeterSuh-Q3, 2023-06-11:
r8125 和 r8169 是否一起使用？

limitget, 2023-06-11:
23.6.4 恢复正常编译，但是无法启动！

limitget, 2023-06-11:
[image]

wjz304, 2023-06-11:
> r8125 和 r8169 是否一起使用？

针对 r81** 的驱动 有啥研究吗？

PeterSuh-Q3, 2023-06-11:
> > r8125 和 r8169 是否一起使用？
> 
> 针对 r81** 的驱动 有啥研究吗？

在 23.6.3 版本中同时使用 r8125 和 r8169 的用户报告了问题。
23.6.2版本说没问题，就又回到了这个版本。

wjz304, 2023-06-11:
> > > r8125 和 r8169 是否一起使用？
> > 
> > 
> > 针对 r81** 的驱动 有啥研究吗？
> 
> 在 23.6.3 版本中同时使用 r8125 和 r8169 的用户报告了问题。 23.6.2版本说没问题，就又回到了这个版本。

那应该是 realtek.ko 的问题吧，realtek.ko 会提高r8169的兼容性。

wjz304, 2023-06-11:
> > > > r8125 和 r8169 是否一起使用？
> > > 
> > > 
> > > 针对 r81** 的驱动 有啥研究吗？
> > 
> > 
> > 在 23.6.3 版本中同时使用 r8125 和 r8169 的用户报告了问题。 23.6.2版本说没问题，就又回到了这个版本。
> 
> 那应该是 realtek.ko 的问题吧，realtek.ko 会提高r8169的兼容性。

realtek 的网卡太难搞了，各种兼容性问题。
[image]

wjz304, 2023-06-11:
> [image] 我使用的是GXNAS版本，就能正常安装最新版本

用着吧

wjz304, 2023-06-14:
https://github.com/wjz304/arpl-i18n/issues/71

closed 2023-06-14 by wjz304