RROrg/rr #83: esxi 6.7 "Loader disk neither USB or DoM"

tjpicole, 2023-06-12:
你好，我是惠普 Microserver GEN8的，安装的esxi6.7版本。目前是用arpl引导安装的7.1.1的，这个版本用了蛮久了，这一波也开机100多天没怎么管它。
看到最近有7.2的版本发布，也有点心心念念，想看看有啥提升。然后我就新建一虚拟机，以前装过好多次原版arpl，很多过程是熟悉的。不过用arpl-i18n的23.6.4版本转换好，新建了虚拟机后开机，跳出来这个提示，Loader disk neither USB or DoM，有点不明白，这个难道不能虚拟机安装么，硬盘配置我也是最简单方式，因为先想着安装测试，所以就是一个SSD的256G固态硬盘，上传了引导文件，额外建立了一个200G的硬盘空间，然后启动后就这个提示，进不了编译界面，不知道下一步怎么继续，谢谢！
虚拟机设置：2CPU，4G，引导盘，200G硬盘，网卡E1000e；

tjpicole, 2023-06-12:
[image]
[image]

tjpicole, 2023-06-12:
好像200G的硬盘不在里面，是硬盘太小不够安装？还是硬盘没认出来，但是引导文件我也是放在这个硬盘上的呀，引导是能引导的么，尴尬

wjz304, 2023-06-12:
sata

closed 2023-06-12 by wjz304