RROrg/rr #84: Realtek feedback

wjz304, 2023-06-12:
Please explain all network cards, availability, issues,

limitget, 2023-06-12:
[image]
不可用

naoki66, 2023-06-14:
Arpl version:[arpl-i18n-23.6.4-Realtek-test.img.zip](https://github.com/wjz304/arpl-i18n/releases/download/23.6.4/arpl-i18n-23.6.4-Realtek-test.img.zip)
DSM model:DS918+
DSM version:64561
--------------------------------------
lspci
00:1f.6 Ethernet controller: Intel Corporation Ethernet Connection (7) I219-V (rev 10)
01:00.0 PCI bridge: ASMedia Technology Inc. ASM1812 6-Port PCIe x4 Gen2 Packet Switch (rev 01)
02:00.0 PCI bridge: ASMedia Technology Inc. ASM1812 6-Port PCIe x4 Gen2 Packet Switch (rev 01)
02:02.0 PCI bridge: ASMedia Technology Inc. ASM1812 6-Port PCIe x4 Gen2 Packet Switch (rev 01)
02:03.0 PCI bridge: ASMedia Technology Inc. ASM1812 6-Port PCIe x4 Gen2 Packet Switch (rev 01)
02:08.0 PCI bridge: ASMedia Technology Inc. ASM1812 6-Port PCIe x4 Gen2 Packet Switch (rev 01)
02:0a.0 PCI bridge: ASMedia Technology Inc. ASM1812 6-Port PCIe x4 Gen2 Packet Switch (rev 01)
02:0b.0 PCI bridge: ASMedia Technology Inc. ASM1812 6-Port PCIe x4 Gen2 Packet Switch (rev 01)
04:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
05:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
07:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
08:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)

---------------------------------------
lspci -v

08:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
        Subsystem: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller
        Flags: bus master, fast devsel, latency 0, IRQ 19
        I/O ports at 3000 [size=256]
        Memory at 8f810000 (64-bit, non-prefetchable) [size=64K]
        Memory at 8f820000 (64-bit, non-prefetchable) [size=16K]
        Expansion ROM at 8f800000 [disabled] [size=64K]
        Capabilities: [40] Power Management version 3
        Capabilities: [50] MSI: Enable- Count=1/1 Maskable+ 64bit+
        Capabilities: [70] Express Endpoint, MSI 01
        Capabilities: [b0] MSI-X: Enable+ Count=32 Masked-
        Capabilities: [d0] Vital Product Data
        Capabilities: [100] Advanced Error Reporting
        Capabilities: [148] Virtual Channel
        Capabilities: [168] Device Serial Number 01-00-00-00-68-4c-e0-00
        Capabilities: [178] Transaction Processing Hints
        Capabilities: [204] Latency Tolerance Reporting
        Capabilities: [20c] L1 PM Substates
        Capabilities: [21c] Vendor Specific Information: ID=0002 Rev=4 Len=100 <?>
        Kernel driver in use: r8125
        Kernel modules: r8125
-----------------------------------------------------------------
Eth0 model:r8125
PID&VID:vid=0x090c，pid=0x1000
available:OK
Eth1 model:r8125
available:OK
Eth2 model:r8125
available:OK
Eth3 model:r8125
available:OK
Eth4 model:e1000e
available:OK
--------------------------------------------------------------
终于正常了，非常感谢！但是请问HddHotplug=0这个怎么修改呢？

wjz304, 2023-06-14:
> 请问HddHotplug=0这个怎么修改呢？

直接再cmdline 里修改就可以了，(注意key value 中如果有空格加双引号)

wjz304, 2023-06-14:
> > 请问HddHotplug=0这个怎么修改呢？
> 
> 直接再cmdline 里修改就可以了，(注意key value 中如果有空格加双引号)

TheChief5055, 2023-06-16:
RTL8111GR (Asrock J3355B-ITX) — no LAN after DSM boot since 23.6.2. Direct boot / text LKM do nothing.

wjz304, 2023-06-16:
Although I said I will no longer compile r8168, the reality is that r8168 still exists in the system, which is built-in in the DSM series

wjz304, 2023-06-16:
get the pid&vid

closed 2023-07-03 by wjz304