RROrg/rr #85: BUG J4125 RTL8125物理机掉线很频繁

yanghanbin, 2023-06-12:
Install ENV:
 * motherboard:
 * CPU:J4125
 * NIC:

Arpl version:23.6.4

DSM model:920+
DSM version:DSM7.2-64570

Issue:

logs:

23.6.4，已经移除了R8169模块，重新编译启动。
如题，运行一段时间后，几个小时，莫名就再也连不上了，在路由器里面能看到机器，但是连不上了，断电重启后正常
======================
另外，N5105的物理机，/usr/bin/arpl-reboot.sh "config"正常升级到23.6.4后，正常启动，在升级里面升级到7.2-64570，连接提示找不到网页了，1-2分钟后，机器彻底掉线。

wjz304, 2023-06-12:
6.3之前的版本呢？

wjz304, 2023-06-12:
I always forget about this.

wjz304, 2023-06-12:
I'm a bit busy today. My PVE broke and I'm currently reinstalling it 
[image]

AuxXxilium, 2023-06-12:
no problem for me, but i think it will help you to find the issue.

yanghanbin, 2023-06-13:
> 开了节能吧？

机器设置一般没动，昨晚尝试关闭uefi，直接启动项都没有了，今天早上还原了，然后当前已经正常运行2个小时了，继续观察

closed 2023-06-13 by yanghanbin