RROrg/rr #87: 关于powersched 扩展的问题

dominicx, 2023-06-13:
版本：23.6.2

pocopico/rp-ext 中powersched  依赖rtc-cmos.ko。但是如果没有安装这个扩展，/usr/sbin/powersched 将不存在
/etc/crontab文件中
会有很多 行 *       *       *       *       *       root    /usr/sbin/powersched

这样会报错。bash_err.log中会有很多错误。

望修正。非常感谢这项给大家带来方便的的工作。

[image]
[image]
[image]

wjz304, 2023-06-13:
dsm 版本？

wjz304, 2023-06-13:
[image]
现在这些架构下是有的

dominicx, 2023-06-14:
> [image] 现在这些架构下是有的
dsm 918+ 7.2-64570

谢谢！

wjz304, 2023-06-14:
检查 modules下吧，应该有的

dominicx, 2023-06-14:
> 检查 modules下吧，应该有的

我手工下载了一个，也删除了crontab里面的命令。powersched应该是选择编译才有这个的。应该会添加一个service和一个timer。

wjz304, 2023-06-18:
然后呢？

zhudan, 2023-06-21:
相同问题，一对bash_err

wjz304, 2023-06-24:
长时间未回复，如新版有问题，请重新创建issue

closed 2023-06-24 by wjz304