RROrg/rr #89: I226V网卡build loader后启动失联

Hawley-hy, 2023-06-13:
Install ENV:
 * motherboard:
 * CPU:N6005
 * NIC:

Arpl version: arpl-i18n-23.6.4.img

DSM model: DSM918+
DSM version: DSM 7.2

Issue:
I226V网卡build loader后启动失联, 路由器上看不到IP。
logs:

...

Hawley-hy, 2023-06-13:
启动画面 eth0 ~ eth3 全部error

yanghanbin, 2023-06-14:
好像是编译后，启动，实际上不会启动，手动重启下，好像就正常了
我物理机接显示器，也以为编译后，点启动就没问题了，然后去看显示器，界面一直没变化，就手动重启了下，恢复正常

wunihehuan, 2023-06-14:
报警吧，老是失联也不是办法

Hawley-hy, 2023-06-14:
@yanghanbin 
重启了也是这样，在启动菜单引导后等2分钟，网卡黄色的灯会熄灭。然后路由器上彻底没有IP了。

Hawley-hy, 2023-06-14:
@wunihehuan 
你有b吧，有b吃点药去。别在这bb

wunihehuan, 2023-06-14:
> @wunihehuan 
> 你有b吧，有b吃点药去。别在这bb

你才有病呢

wjz304, 2023-06-14:
https://qun.qq.com/qqweb/qunpro/share?_wv=3&_wwv=128&appChannel=share&inviteCode=20jx8dPsU2z&contentID=1m4NKs&businessType=2&from=181174&shareSource=5&biz=ka

Hawley-hy, 2023-06-14:
@wjz304 
感谢大神

closed 2023-06-15 by Hawley-hy