RROrg/rr #90: BUG: NIC e1000 vs e1000e wrong driver loaded

naeemsufian, 2023-06-15:
Install ENV:
 * motherboard: HP Z240 SFF Workstation 2nd HP800G1
 * CPU: corei5
 * NIC:Both Machines has e1000 Network

Arpl version:23.6.4

DSM model: ds918+
DSM version: 9.2.0-64570

Issue: e1000e driver loaded instead of e1000 

logs: Not getting ip after build

...

wjz304, 2023-06-15:
Can you get the pid&vid of the NIC?

naeemsufian, 2023-06-15:
Yes no problem with video and pid

naeemsufian, 2023-06-16:
> Can you get the pid&vid of the NIC?

HP800G1
NIC e1000
[image]

wjz304, 2023-06-16:
```
$ modinfo e1000.ko | grep alias:
alias:          pci:v00008086d00002E6Esv*sd*bc*sc*i*
alias:          pci:v00008086d000010B5sv*sd*bc*sc*i*
alias:          pci:v00008086d00001099sv*sd*bc*sc*i*
alias:          pci:v00008086d0000108Asv*sd*bc*sc*i*
alias:          pci:v00008086d0000107Csv*sd*bc*sc*i*
alias:          pci:v00008086d0000107Bsv*sd*bc*sc*i*
alias:          pci:v00008086d0000107Asv*sd*bc*sc*i*
alias:          pci:v00008086d00001079sv*sd*bc*sc*i*
alias:          pci:v00008086d00001078sv*sd*bc*sc*i*
alias:          pci:v00008086d00001077sv*sd*bc*sc*i*
alias:          pci:v00008086d00001076sv*sd*bc*sc*i*
alias:          pci:v00008086d00001075sv*sd*bc*sc*i*
alias:          pci:v00008086d00001028sv*sd*bc*sc*i*
alias:          pci:v00008086d00001027sv*sd*bc*sc*i*
alias:          pci:v00008086d00001026sv*sd*bc*sc*i*
alias:          pci:v00008086d0000101Esv*sd*bc*sc*i*
alias:          pci:v00008086d0000101Dsv*sd*bc*sc*i*
alias:          pci:v00008086d0000101Asv*sd*bc*sc*i*
alias:          pci:v00008086d00001019sv*sd*bc*sc*i*
alias:          pci:v00008086d00001018sv*sd*bc*sc*i*
alias:          pci:v00008086d00001017sv*sd*bc*sc*i*
alias:          pci:v00008086d00001016sv*sd*bc*sc*i*
alias:          pci:v00008086d00001015sv*sd*bc*sc*i*
alias:          pci:v00008086d00001014sv*sd*bc*sc*i*
alias:          pci:v00008086d00001013sv*sd*bc*sc*i*
alias:          pci:v00008086d00001012sv*sd*bc*sc*i*
alias:          pci:v00008086d00001011sv*sd*bc*sc*i*
alias:          pci:v00008086d00001010sv*sd*bc*sc*i*
alias:          pci:v00008086d0000100Fsv*sd*bc*sc*i*
alias:          pci:v00008086d0000100Esv*sd*bc*sc*i*
alias:          pci:v00008086d0000100Dsv*sd*bc*sc*i*
alias:          pci:v00008086d0000100Csv*sd*bc*sc*i*
alias:          pci:v00008086d00001009sv*sd*bc*sc*i*
alias:          pci:v00008086d00001008sv*sd*bc*sc*i*
alias:          pci:v00008086d00001004sv*sd*bc*sc*i*
alias:          pci:v00008086d00001001sv*sd*bc*sc*i*
alias:          pci:v00008086d00001000sv*sd*bc*sc*i*

$ modinfo e1000e.ko | grep alias:
alias:          pci:v00008086d00001A1Dsv*sd*bc*sc*i*
alias:          pci:v00008086d00001A1Csv*sd*bc*sc*i*
alias:          pci:v00008086d00001A1Fsv*sd*bc*sc*i*
alias:          pci:v00008086d00001A1Esv*sd*bc*sc*i*
alias:          pci:v00008086d000015F5sv*sd*bc*sc*i*
alias:          pci:v00008086d000015F4sv*sd*bc*sc*i*
alias:          pci:v00008086d000015FAsv*sd*bc*sc*i*
alias:          pci:v00008086d000015F9sv*sd*bc*sc*i*
alias:          pci:v00008086d000015FCsv*sd*bc*sc*i*
alias:          pci:v00008086d000015FBsv*sd*bc*sc*i*
alias:          pci:v00008086d00000D4Dsv*sd*bc*sc*i*
alias:          pci:v00008086d00000D4Csv*sd*bc*sc*i*
alias:          pci:v00008086d00000D4Fsv*sd*bc*sc*i*
alias:          pci:v00008086d00000D4Esv*sd*bc*sc*i*
alias:          pci:v00008086d000015E2sv*sd*bc*sc*i*
alias:          pci:v00008086d000015E1sv*sd*bc*sc*i*
alias:          pci:v00008086d000015E0sv*sd*bc*sc*i*
alias:          pci:v00008086d000015DFsv*sd*bc*sc*i*
alias:          pci:v00008086d000015BCsv*sd*bc*sc*i*
alias:          pci:v00008086d000015BBsv*sd*bc*sc*i*
alias:          pci:v00008086d000015BEsv*sd*bc*sc*i*
alias:          pci:v00008086d000015BDsv*sd*bc*sc*i*
alias:          pci:v00008086d00000D55sv*sd*bc*sc*i*
alias:          pci:v00008086d00000D53sv*sd*bc*sc*i*
alias:          pci:v00008086d000015D6sv*sd*bc*sc*i*
alias:          pci:v00008086d000015E3sv*sd*bc*sc*i*
alias:          pci:v00008086d000015D8sv*sd*bc*sc*i*
alias:          pci:v00008086d000015D7sv*sd*bc*sc*i*
alias:          pci:v00008086d000015B9sv*sd*bc*sc*i*
[... 40 lines cut ...]
alias:          pci:v00008086d0000104Csv*sd*bc*sc*i*
alias:          pci:v00008086d000010BBsv*sd*bc*sc*i*
alias:          pci:v00008086d00001098sv*sd*bc*sc*i*
alias:          pci:v00008086d000010BAsv*sd*bc*sc*i*
alias:          pci:v00008086d00001096sv*sd*bc*sc*i*
alias:          pci:v00008086d0000150Csv*sd*bc*sc*i*
alias:          pci:v00008086d000010F6sv*sd*bc*sc*i*
alias:          pci:v00008086d000010D3sv*sd*bc*sc*i*
alias:          pci:v00008086d0000109Asv*sd*bc*sc*i*
alias:          pci:v00008086d0000108Csv*sd*bc*sc*i*
alias:          pci:v00008086d0000108Bsv*sd*bc*sc*i*
alias:          pci:v00008086d0000107Fsv*sd*bc*sc*i*
alias:          pci:v00008086d0000107Esv*sd*bc*sc*i*
alias:          pci:v00008086d0000107Dsv*sd*bc*sc*i*
alias:          pci:v00008086d000010B9sv*sd*bc*sc*i*
alias:          pci:v00008086d000010D5sv*sd*bc*sc*i*
alias:          pci:v00008086d000010DAsv*sd*bc*sc*i*
alias:          pci:v00008086d000010D9sv*sd*bc*sc*i*
alias:          pci:v00008086d00001060sv*sd*bc*sc*i*
alias:          pci:v00008086d000010A5sv*sd*bc*sc*i*
alias:          pci:v00008086d000010BCsv*sd*bc*sc*i*
alias:          pci:v00008086d000010A4sv*sd*bc*sc*i*
alias:          pci:v00008086d0000105Fsv*sd*bc*sc*i*
alias:          pci:v00008086d0000105Esv*sd*bc*sc*i*
```

wjz304, 2023-06-16:
Perhaps it is really e1000e?, The driver of e1000 does not support it

naeemsufian, 2023-06-16:
> pci:v00008086d0000153Asv*sd*bc*sc*i*            8086&153A

yes i got it... so i reburn Arpl to another usb still, again got error no luck. Then i  update system bios  and its start working well now. 
many many thanks for the time and help.

naeemsufian, 2023-06-16:
i update system bios and its start working well now.
First time i am using this arpl-i18n version Arpl 23.6.4 loader and its working well and smooth.
Highly appreciate and thanks to the developer  wjz304.

closed 2023-06-16 by naeemsufian