RROrg/rr #94: BUG

superyijia, 2023-06-17:
Install ENV:26.5/26.4
 * motherboard:
 * CPU:i5 3317U
 * NIC:rtl 8168/r8152

Arpl version:26.5/26.4

DSM model:920+
DSM version:64570

Issue:安装DSM.PAT时，提示文件无法安装，已经毁损

logs:
[](url)

...
[image]

wjz304, 2023-06-18:
三代装不了吧，我没有三代，具体表现是啥 我不知道

superyijia, 2023-06-18:
感谢大佬回复，可能三代真的装不了，我周末把所有的版本都试了一遍。都不可以。Intel CPU要几代开始可以使用丫？感谢指点

gmy771810930, 2023-06-20:
> 感謝大佬回復，可能三代真的裝不了，我週末把所有的版本都試了一遍。都不可以。Intel CPU要幾代開始可以使用丫？感謝指點

一般是4-10，11及其以上核显用不了

cyngu5767, 2023-06-20:
> 感谢大佬回复，可能三代真的装不了，我周末把所有的版本都试了一遍。都不可以。Intel CPU要几代开始可以使用丫？感谢指点

Intel4代以后才可以安装DS918/920，四代以前的CPU可以考虑安装DS3617/3622。

xuehrs, 2023-06-22:
j1900可否，
发现gxnasblog ds918 7.2可以用

javaMumu, 2023-06-27:
DSM model:920+/918+
DSM version:64570

javaMumu, 2023-06-27:
CPU: N5105
Arpl version:26.7
DSM model:920+
DSM version:64570
无法安装DSM，提示文件可能损毁

superyijia, 2023-06-27:
I have found the cause of this problem, the hard drive has a bad channel.
After I replaced the hard drive, everything was fine.

csvwolf, 2023-07-01:
intel j3455 也遇到了这个问题

misteruly, 2023-07-06:
我用虚拟机也装不了啊

cpymax, 2023-07-22:
1037U 用原版arpl可以安装7.1，用23.7安装7.1 7.2 都会提示损毁，正在尝试以前的版本

原版的话没有在线安装，不知道是不是和这个有关系

Erazul, 2023-07-30:
BOOTING: [UEFI] [USB flashdisk]
型号: DS918+
版本: 7.2 7.1.1
LKM: prod
CPU: i5-4590
MEM: 16GB

遇到了一样的问题，手动上传群晖固件，联网状态下到百分之55 提示文件损毁，未联网状态下百分之56 提示文件损毁。
之前arpl 7.1.1安装正常但是忘记用的是哪个版本了.

wjz304, 2023-07-30:
安装盘 多大？

cpymax, 2023-07-30:
> 安装盘 多大？

有一个是16g，另一个没注意，需要大容量的才行吗？

misteruly, 2023-07-31:
安装盘120G
 

misteruly
***@***.***

Erazul, 2023-07-31:
> 安装盘 多大？

引导U盘 试过 64g 和 2g 的
安装盘 256g 三星sata固态 + 6TB 希捷银河机械硬盘

Erazul, 2023-08-05:
> BOOTING: [UEFI] [USB flashdisk] 型号: DS918+ 版本: 7.2 7.1.1 LKM: prod CPU: i5-4590 MEM: 16GB
> 
> 遇到了一样的问题，手动上传群晖固件，联网状态下到百分之55 提示文件损毁，未联网状态下百分之56 提示文件损毁。 之前arpl 7.1.1安装正常但是忘记用的是哪个版本了.

我后来用了另一种方法搞定了：
先在本机上安装esxi到U盘上
直通sata控制器
新建一个虚拟机加上arpl引导安装成功 7.2 up3
通过群晖给sata上的硬盘创建池
拔掉esxi虚拟机u盘，制作另一个arpl引导U盘插上开机
“高级设置——恢复已安装DSM系统”
问题解决！

wjz304, 2023-08-22:
暂时关闭

closed 2023-08-22 by wjz304

lighcen, 2023-11-20:
> 一样遇到了这个问题
> 
> 尝试了下人工改了下 grub.cfg，在插上去安装成功

要怎么修改呢？