RROrg/rr #95: R8125B安装完引导后可以查询到但是连接不上

Desmond56, 2023-06-18:
Install ENV:
 * motherboard:
 * CPU:
 * NIC:

Arpl version:

DSM model:
DSM version:

Issue:

logs:

...

Desmond56, 2023-06-18:
[image]
[image]

wjz304, 2023-06-18:
取 pid vid

wjz304, 2023-06-18:
web 有缓存，以路由分配为准。

https://qun.qq.com/qqweb/qunpro/share?_wv=3&_wwv=128&appChannel=share&inviteCode=20jx8dPsU2z&contentID=1m4NKs&businessType=2&from=181174&shareSource=5&biz=ka

Desmond56, 2023-06-18:
> 

这俩不是自动设置的吗

Desmond56, 2023-06-18:
有文档频道地址吗，有没有编译的教程，我看看是不是我编译差东西了

Desmond56, 2023-06-18:
> web 有缓存，以路由分配为准。
> 
> https://qun.qq.com/qqweb/qunpro/share?_wv=3&_wwv=128&appChannel=share&inviteCode=20jx8dPsU2z&contentID=1m4NKs&businessType=2&from=181174&shareSource=5&biz=ka

还有就是，我在我的路由器里看到了两个ip

wjz304, 2023-06-18:
你说网卡的问题 要提供网卡的vid pid啊，

Desmond56, 2023-06-21:
> 你说网卡的问题要提供网卡的vid pid啊，

实在是没找到怎么查看网卡vid,pid方法

wjz304, 2023-06-21:
arpl shell 下 lspci -nn

Desmond56, 2023-06-22:
> arpl shell 下 lspci -nn

[image]

wjz304, 2023-06-22:
https://qun.qq.com/qqweb/qunpro/share?_wv=3&_wwv=128&appChannel=share&inviteCode=20jx8dPsU2z&contentID=1m4NKs&businessType=2&from=181174&shareSource=5&biz=ka

Desmond56, 2023-06-22:
启动后无法连接怎么返回配置引导那个状态呢, 我现在想切换dev只能只能重新做引导

Desmond56, 2023-06-22:
已解决,蟹蟹

closed 2023-06-22 by Desmond56