RROrg/rr #98: BUG

joewen85, 2023-06-25:
Arpl version: 23.6.7

DSM model: dva3221
DSM version: 64570

Issue:
使用x520 sr-iov虚拟网卡.  config能通过分配的ip网页设置. 一切均正常.  设置完后,使用find.synology.com查找设备发现ip不对
[image]

[image]

查了一下: https://www.openos.org/threads/1554/page-6
[image]
不清楚是否这问题导致

wjz304, 2023-06-25:
esxi ？

wjz304, 2023-06-25:
https://qun.qq.com/qqweb/qunpro/share?_wv=3&_wwv=128&appChannel=share&inviteCode=20jx8dPsU2z&contentID=1m4NKs&businessType=2&from=181174&shareSource=5&biz=ka

joewen85, 2023-06-25:
使用dev模式会 pts/0无权限
[image]

wjz304, 2023-06-25:
不用管那个无权限，

joewen85, 2023-06-26:
> 

昨晚也测试了,无论用什么版本.esxi虚拟的网卡 sr-iov, vmxnet3都会出现无法访情况. sr-iov安装阶段已找不到设备. vmxnet3安装完后进入系统找不到网卡了. 请问上文所说的debug日志要重定向那个文件到串口呢? /var/log/message看了一下err只有samba service的错误

wjz304, 2023-06-26:
lsmod

dmesg | grep vmxnet3

物理网卡是什么？ 再 取 dmesg | grep <对应的虚拟网卡驱动>

joewen85, 2023-06-26:
物理网卡 82599 x520双口的
[image]

wjz304, 2023-06-26:
ifconfig

ls -ld /sys/class/net/*/device/driver

joewen85, 2023-06-26:
> ifconfig
> 
> ls -ld /sys/class/net/*/device/driver

[image]
[image]

wjz304, 2023-06-26:
.... driver

joewen85, 2023-06-26:
> .... driver

没有driver

wjz304, 2023-06-26:
额，你这是在 arpl下啊...  看 dsm下的

joewen85, 2023-06-26:
> 额，你这是在 arpl下啊... 看 dsm下的

dsm请问怎么连ssh进去看呢?

wjz304, 2023-06-26:
[image]
直通 + vmxnet3 + ixgbevf

wjz304, 2023-06-26:
用能连上的设备连上 去看不好使的设备为什么不好使啊

joewen85, 2023-06-26:
> [image]
> 直通 + vmxnet3 + ixgbevf

只要接入vmxnet3  ixgbevf其中一个,连直通的也进不去了.

wjz304, 2023-06-26:
esxi？

joewen85, 2023-06-26:
> esxi？

对的. esxi中使用sr-iov

wjz304, 2023-06-26:
在 arpl下
`sed -i 's/ttyS0/ttyS1/g' /opt/arpl/boot.sh`
然后再菜单中持久化一下
虚拟机田间串口日志到文件。

joewen85, 2023-06-26:
好像也没有啥错误
[dsm_debug.log](https://github.com/wjz304/arpl-i18n/files/11866762/dsm_debug.log)

wjz304, 2023-06-26:
ttyS0 被关掉了，看不到后面的日志。么的办法了，

joewen85, 2023-06-26:
> ttyS0 被关掉了，看不到后面的日志。么的办法了，

不是改为ttyS1吗?

joewen85, 2023-06-26:
确认爱快、win10直接加直通的sr-iov虚拟网卡是稳定正常使用的. 可能问题还是在群晖上

wjz304, 2023-06-26:
是的，这个驱动是抓的kvmx64平台的，可能会不兼容。但是看不到日志都白扯，都是猜测。

wjz304, 2023-06-26:
如果是pve 就有串口了

squaly1, 2023-06-27:
同一问题esxi里面的sriov网卡无法访问ip，双网卡其中一个vmxnet3也不行，必须去掉sriov才能访问ip，原版arpl没这问题，可惜不支持7.2

joewen85, 2023-06-27:
> 同一问题esxi里面的sriov网卡无法访问ip，双网卡其中一个vmxnet3也不行，必须去掉sriov才能访问ip，原版arpl没这问题，可惜不支持7.2

是指那个?我试试

squaly1, 2023-06-27:
> > 同一问题esxi里面的sriov网卡无法访问ip，双网卡其中一个vmxnet3也不行，必须去掉sriov才能访问ip，原版arpl没这问题，可惜不支持7.2
> 
> [image] 是这个吗? 这个支持7.2

是这个，但是不支持7.2啊，软件版本根本没有7.2选项，好奇怪

joewen85, 2023-06-27:
确实用v1.1-beta2a可以走安装向导了,但安装时跑到8%就报无法格式化了
[image]
[image]

请问这问题有解吗? 还是看看咋修你的项目呢?

joewen85, 2023-06-27:
> > > 同一问题esxi里面的sriov网卡无法访问ip，双网卡其中一个vmxnet3也不行，必须去掉sriov才能访问ip，原版arpl没这问题，可惜不支持7.2
> > 
> > 
> > [image] 是这个吗? 这个支持7.2
> 
> 是这个，但是不支持7.2啊，软件版本根本没有7.2选项，好奇怪

现在你是怎样方案呢?

squaly1, 2023-06-27:
> > > > 同一问题esxi里面的sriov网卡无法访问ip，双网卡其中一个vmxnet3也不行，必须去掉sriov才能访问ip，原版arpl没这问题，可惜不支持7.2
> > > 
> > > 
> > > [image] 是这个吗? 这个支持7.2
> > 
> > 是这个，但是不支持7.2啊，软件版本根本没有7.2选项，好奇怪
> 
> 现在你是怎样方案呢? 需要dva那8个 nvr的授权,又能用上万兆就行了

我现在用虚拟网卡安装的，我都想问你有没有其他办法，所有虚拟机都是sriov的，只有群晖是虚拟网卡，内网连接不是很稳定，真郁闷，只能等你把完整日志给作者看怎么解决了。

joewen85, 2023-06-27:
> > > > > 同一问题esxi里面的sriov网卡无法访问ip，双网卡其中一个vmxnet3也不行，必须去掉sriov才能访问ip，原版arpl没这问题，可惜不支持7.2
> > > > 
> > > > 
> > > > [image] 是这个吗? 这个支持7.2
> > > 
> > > 
> > > 是这个，但是不支持7.2啊，软件版本根本没有7.2选项，好奇怪
> > 
> > 
> > 现在你是怎样方案呢? 需要dva那8个 nvr的授权,又能用上万兆就行了
> 
> 我现在用虚拟网卡安装的，我都想问你有没有其他办法，所有虚拟机都是sriov的，只有群晖是虚拟网卡，内网连接不是很稳定，真郁闷，只能等你把完整日志给作者看怎么解决了。

作者说被关闭日志输出无法看到往后的日志. 用v1.1-beata2a的引导 dva3221能安装吗? 我这边安装8%的时候报错无法格式化.请问你用的是什么型号呢?

squaly1, 2023-06-27:
> 作者说被关闭日志输出无法看到往后的日志. 用v1.1-beata2a的引导 dva3221能安装吗? 我这边安装8%的时候报错无法格式化.请问你用的是什么型号呢?

dva3221是什么？1.1那个引导的菜单完全没有7.2，7.1倒是正常。
https://github.com/PeterSuh-Q3/tinycore-redpill/releases/tag/v0.9.4.3-2
用红丸好像可以安装，你试试，如果可以告诉我，困了，今晚不折腾了。

joewen85, 2023-06-27:
> [image]
> > > > > > > 同一问题esxi里面的sriov网卡无法访问ip，双网卡其中一个vmxnet3也不行，必须去掉sriov才能访问ip，原版arpl没这问题，可惜不支持7.2
> > > > > > 
> > > > > > 
> > > > > > [image] 是这个吗? 这个支持7.2
> > > > > 
> > > > > 
> > > > > 是这个，但是不支持7.2啊，软件版本根本没有7.2选项，好奇怪
> > > > 
> > > > 
> > > > 现在你是怎样方案呢? 需要dva那8个 nvr的授权,又能用上万兆就行了
> > > 
> > > 
> > > 我现在用虚拟网卡安装的，我都想问你有没有其他办法，所有虚拟机都是sriov的，只有群晖是虚拟网卡，内网连接不是很稳定，真郁闷，只能等你把完整日志给作者看怎么解决了。
> > 
> > 
> > 作者说被关闭日志输出无法看到往后的日志. 用v1.1-beata2a的引导 dva3221能安装吗? 我这边安装8%的时候报错无法格式化.请问你用的是什么型号呢?
> 
> dva3221是什么？1.1那个引导的菜单完全没有7.2，7.1倒是正常。 v1.1-beata2a你可以正常显示编译7.2吗？ https://github.com/PeterSuh-Q3/tinycore-redpill/releases/tag/v0.9.4.3-2 用红丸好像可以安装，你试试，如果可以告诉我，困了，今晚不折腾了。

dva3221是synoloyg型号 v1.1是没有7.2  测试了使用v1.1 型号dva1622正常了

joewen85, 2023-06-27:
> 如果是pve 就有串口了

sr-iov有个特性的就是每次重启设备mac地址都会不一样,与这有没有关系呢?

wjz304, 2023-06-27:
更新模块 23.6.12 试试吧，没有环境，我测不了

squaly1, 2023-06-27:
> > 如果是pve 就有串口了
> 
> sr-iov有个特性的就是每次重启设备mac地址都会不一样,与这有没有关系呢?
[image]

[image]

型号设置了dva1622还是没有看见7.2啊？你说怎么设置的？截图看看，dva1622有没有核显？我现在用920，换来这个型号使用都一样吗？有没有原生支持m2做储存空间？

squaly1, 2023-06-28:
> 更新模块 23.6.12 试试吧，没有环境，我测不了

只看到23.6.7，在esxi设置sriov为唯一的网卡，可以进入设置界面，并进行设置，但是一但设置完重启后就ping不到ip了

joewen85, 2023-06-28:
> > > 如果是pve 就有串口了
> > 
> > 
> > sr-iov有个特性的就是每次重启设备mac地址都会不一样,与这有没有关系呢?
> > [image]
> 
> [image]
> 
> 型号设置了dva1622还是没有看见7.2啊？你说怎么设置的？截图看看，dva1622有没有核显？我现在用920，换来这个型号使用都一样吗？有没有原生支持m2做储存空间？

我是用v1.1的 型号dva1622 7.1版本系统. 我的需求可实现了.  1622有没有核显,m2是否支持没测试.

joewen85, 2023-06-28:
> > 更新模块 23.6.12 试试吧，没有环境，我测不了
> 
> 只看到23.6.7，在esxi设置sriov为唯一的网卡，可以进入设置界面，并进行设置，但是一但设置完重启后就ping不到ip了

那应该是群晖7.2绑了mac了. 按照sr-iov的特性来推断. 备份完数据我降级用回7.1试试

squaly1, 2023-06-28:
> > > > 如果是pve 就有串口了
> > > 
> > > 
> > > sr-iov有个特性的就是每次重启设备mac地址都会不一样,与这有没有关系呢?
> > > [image]
> > 
> > 
> > [image]
> > 型号设置了dva1622还是没有看见7.2啊？你说怎么设置的？截图看看，dva1622有没有核显？我现在用920，换来这个型号使用都一样吗？有没有原生支持m2做储存空间？
> 
> 我是用v1.1的 型号dva1622 7.1版本系统. 我的需求可实现了. 1622有没有核显,m2是否支持没测试.

42962是7.11啊 64561以上才是7.2

joewen85, 2023-06-28:
> > > > > 如果是pve 就有串口了
> > > > 
> > > > 
> > > > sr-iov有个特性的就是每次重启设备mac地址都会不一样,与这有没有关系呢?
> > > > [image]
> > > 
> > > 
> > > [image]
> > > 型号设置了dva1622还是没有看见7.2啊？你说怎么设置的？截图看看，dva1622有没有核显？我现在用920，换来这个型号使用都一样吗？有没有原生支持m2做储存空间？
> > 
> > 
> > 我是用v1.1的 型号dva1622 7.1版本系统. 我的需求可实现了. 1622有没有核显,m2是否支持没测试.
> 
> 42962是7.11啊 64561以上才是7.2

应该是群晖7.2系统改机制了. 这个看看以后能不能破吧

wjz304, 2023-06-28:
> > 更新模块 23.6.12 试试吧，没有环境，我测不了
> 
> 只看到23.6.7，在esxi设置sriov为唯一的网卡，可以进入设置界面，并进行设置，但是一但设置完重启后就ping不到ip了

我说 模块  modules

joewen85, 2023-06-28:
> 更新模块 23.6.12 试试吧，没有环境，我测不了

问题解决了.非常感谢.必须捐助一下
[image]

wjz304, 2023-06-28:
赞赏码让tank给我玩封了，好像明天才会解封
[image]

closed 2023-06-28 by wjz304