RROrg/rr #101: mailplus基本设置无法访问数据库

Catohy, 2023-06-26:
大佬，我之前安装23.5.8的引导升级7.2，一切都正常包括mailplus server，但是我发现一个问题，自从这个版本之后的所有版本，重新安装7.2的系统都无法使用mailplus了，不知道是不是引导中哪里出了小问题。23.5.8之后的所有版本都是报错“基础设置就说数据库无法访问”，其他服务都正常。esxi我当时也换过6，7，8，最后发现居然是引导导致的，不知道大佬能否看一下这个情况，感谢！
[image]

wjz304, 2023-06-30:
已测试多台，没出现你说的情况，自行查找原因吧
[image]

closed 2023-07-02 by wjz304

oemsys, 2024-06-07:
我的也一样，，，，，报错，，，

5UFKEFU, 2024-08-02:
同样的问题

sommir, 2024-11-08:
可能的问题： 配置了系统代理之后就会报错，我关掉之后就解决了。

javierztl, 2024-11-28:
[image]
我也同样打打问题,不知道是不是设定了FRP导致的

buslink, 2025-01-09:
25.1.0同样的问题
[image]

buslink, 2025-01-09:
我锁定到问题了，退出Synology 账户后可以正常使用。登录了账户就不行

bao3, 2025-03-01:
> 我锁定到问题了，退出Synology 账户后可以正常使用。登录了账户就不行

感谢反馈。我退出帐户后也解决了。方法有效。