RROrg/rr #99: I wonder why the checksum of DSM files was switched from SHA256 to MD5.

PeterSuh-Q3, 2023-06-26:
From my point of view, both of these values were set in ARPL's config for each model, and it seems to be to unify unnecessary dual management into one MD5.
Is that right?
Doesn't MD5 have other advantages?

And, the difference between ARPL and TCRP’s core mechanism is that ARPL uses jun mode, so it is handled as a 7.0.1 version file created before encryption, and TCRP, on the contrary, requires reassembly of pat files encrypted after 7.1.0. And the sha256 checksum of the pat file resulting from this recombination is recorded and managed once again in the configuration file.
I also think this part is a bit out of principle.
Perhaps even if TCRP adopts MD5, this part seems unavoidable right now.
In the case of TCRP, only SHA256 is used in the loader build phase.
In the installation phase of the DSM file, MD5 is involved through rss.xml and rss.json.

wjz304, 2023-06-26:
大部分的 sha校验都是为了确定文件的完整性，防止下载出现错误，

pat 下载完成进行一次校验就好了，
没有必要在解压后再次对rd.gz和zimage进行校验。只要pat的解压过程没有报错，他们一般不会出错.
选用md5还是 hash 应该没有区别吧，只是md5 在 rss 中也会用到，所以就只保留了md5.

wjz304, 2023-06-26:
In the future, I may allow users to upload local PAT, allowing them to create some intermediate versions as loader.

wjz304, 2023-06-26:
我觉得 redpill-load 中的 校验也可以去掉一些，太多了，都是 sha256 error，每次修改文件还要想着修改sha256
[image]

PeterSuh-Q3, 2023-06-26:
> 我觉得 redpill-load 中的 校验也可以去掉一些，太多了，都是 sha256 error，每次修改文件还要想着修改sha256 [image]

I am of the same opinion as you. Strict checking is good, but I think it's too much. I tried to modify redpill-load partly, but it was put on hold because I thought I was out of the basic rules of TTG by myself.