RROrg/rr #104: Setting custom SN manually says invalid SN warning

OrpheeGT, 2023-06-28:
Hello,

When I manually set a SN on DVA1622 loader, with UBR trygram, it says invalid SN. (loader ARPL 23.6.8)

Probably because mine begins with 2260, and not part of the listed prefixes :

https://github.com/wjz304/arpl-i18n/blob/main/files/board/arpl/overlayfs/opt/arpl/model-configs/DVA1622.yml#L22
serial:
  prefix:
  - "2030"
  - "2040"
  - "20C0"
  - "2150"
  middle: "UBR"
  suffix: "alpha"

If it is because of prefix not in the list, maybe you should ignore this check any only test trygram (UBR in this case) to validate the SN.

Thanks.

wjz304, 2023-06-28:
Yes, this prompt is actually somewhat misleading,
At first, I wanted to crack the vdsm keygen to achieve a once and for all effect, but later I put it on hold,
Very few people mention it, so I forgot about it.
[image]

wjz304, 2023-06-28:
Using vdsm's sn and mac can make some account related functions available, and you only need to create a vdsm to easily obtain them.

OrpheeGT, 2023-06-28:
> Using vdsm's sn and mac can make some account related functions available, and you only need to create a vdsm to easily obtain them.

I don't know how vdsm SN works, but for DVA series, trygram matching model is mandatory to have Surveillance Station 8 camera licences available.

wjz304, 2023-06-28:
我暂时先把这个提示注释掉吧

closed 2023-06-28 by wjz304