RROrg/rr #107: 不支持ipv6(补充截图，重发）

gexiaopeng, 2023-06-29:
我用虚拟机VMware（版本15）安装的群晖系统 ds3617xs-7.2.0-64570，虚拟机网络模式:NAT(e1000)。在控制面板 网络，网络界面 ipv6是空的，通过 ssh的命令ifconfig 也看不到ipv6地址。 我用其他的引导镜像文件安装，同样的虚拟机设置，群晖系统有ipv6.
截图如下：
１）虚拟机网络 
[image]

２）虚拟机启动画面
[image]

３）群晖系统版本
[image]

４）控制面板网络
[image]

５）群晖系统ssh ifconfig 
[image]

6) 本机cmd ipconfig 
[image]

gexiaopeng, 2023-06-29:
自动回复？ 已经有截图了。

wjz304, 2023-06-29:
[image]

joewen85, 2023-06-29:
[image]
测试可以的, 网卡设置里面ipv6看看是否开启

gexiaopeng, 2023-06-29:
设置 ipv6 为自动以后正常了，非常感谢。 个人建议，最好默认是开启ipv6（自动）。

wjz304, 2023-06-29:
这是系统默认的，引导并不会处理这些。920+ 自带2个网卡，前两个网卡默认就是关的，

closed 2023-06-29 by wjz304