RROrg/rr #108: BUG

BenBenPeng, 2023-06-30:
Model: DS920+
Build: 64570u1
LKM:   prod
CPU:   Intel(R) N100
MEM:   15789 MB
*** Detected 1 network cards, 0 MACs were customized, the rest will use the original MACs. ***
Cmdline:
withefi console=ttyS0,115200n8 earlyprintk earlycon=uart8250,io,0x3f8,115200n8 root=/dev/md0 loglevel=15 log_buf_len=32M syno_hw_version=DS920+ netif_num=1 synoboot2 pid=0x5571 mac1=7c8334b854ee sn=2040SBRCDZLDL HddEnableDynamicPower=1 vid=0x0781 elevator=elevator intel_iommu=igfx_off vender_format_version=2 syno_ttyS1=serial,0x2f8 syno_ttyS0=serial,0x3f8
Detected 1 network cards, Waiting IP.(For reference only)
eth0(r8169): Access http://192.168.199.100:5000 to connect the DSM via web.
Loading DSM kernel...
Booting...

[This interface will not be operational. Please use the http://find.synology.com/ find DSM and connect.]

看上去没问题的呀，重启后就找不到了。

Request timeout for icmp_seq 7401
Request timeout for icmp_seq 7402
Request timeout for icmp_seq 7403
Request timeout for icmp_seq 7404
Request timeout for icmp_seq 7405
ping: sendto: No route to host
Request timeout for icmp_seq 7406
ping: sendto: Host is down
Request timeout for icmp_seq 7407
ping: sendto: Host is down
Request timeout for icmp_seq 7408
ping: sendto: Host is down
Request timeout for icmp_seq 7409
ping: sendto: Host is down
Request timeout for icmp_seq 7410
ping: sendto: Host is down
Request timeout for icmp_seq 7411

wjz304, 2023-06-30:
https://qun.qq.com/qqweb/qunpro/share?_wv=3&_wwv=128&appChannel=share&inviteCode=20jx8dPsU2z&contentID=1m4NKs&businessType=2&from=181174&shareSource=5&biz=ka

closed 2023-07-02 by wjz304

ZY1014, 2023-07-24:
> Model: DS920+ Build: 64570u1 LKM: prod CPU: Intel(R) N100 MEM: 15789 MB *** Detected 1 network cards, 0 MACs were customized, the rest will use the original MACs. *** Cmdline: withefi [...]

解决了吗 我安装完卡在这个界面 无法连接，重启后又要重装